duff2013/Snooze #1: Snooze changes pin function

holgerlembke, 2015-06-09:
I try to use snooze but it changes the behavior of my software.

I have a pump and a switch (usual pullup configuration). If switch is not pressed the pump shall pump. If switch is hold down the pump shall continue pumping for some seconds and then stop. 

It is more or less this:

void setup() {
  pinMode(13,OUTPUT);  // pump
  pinMode(8,INPUT_PULLUP); // switch
}

void loop() {
  static unsigned long ticker;
  if (digitalRead(8)==LOW) {
    if (millis()-ticker>5000) {
      digitalWrite(13,LOW);
	}  
  } else {
    ticker=millis();
    digitalWrite(13,HIGH);
  }	
}

I thought I could send the teensy 3.1 to sleep and wake it up again if button moves... I downloaded current snooze and now it looks like that:

#include <Snooze.h>

SnoozeBlock config;

void setup() {
  pinMode(13,OUTPUT);  // pump
  config.pinMode(8, INPUT_PULLUP, CHANGE);
}

void loop() {
  static unsigned long ticker;
  if (digitalRead(8)==LOW) {
    if (millis()-ticker>5000) {
      digitalWrite(13,LOW);
	}  
  } else {
    ticker=millis();
    digitalWrite(13,HIGH);
  }	
}

I didn't implement the sleep, because it already stopped working; reading the input seems to be always low. 

Any hint? Thanks!

duff2013, 2015-06-09:
```config.pinMode(8, INPUT_PULLUP, CHANGE);``` does not configure the pin until you call one of the sleep functions. You need to add ```pinMode(8,INPUT_PULLUP);``` in the setup function to have that pin configured. This way you can configure pin 8 in your case for run mode operation and have it configured differently for wakeup a source when sleeping.
```c
#include "Snooze.h"

SnoozeBlock config;

void setup() {
  pinMode(13, OUTPUT); // pump
  pinMode(8, INPUT_PULLUP);// Run Mode 
  config.pinMode(8, INPUT_PULLUP, CHANGE);// Sleep Mode
  /********************NOTE!!!!*************************
  * if the above pinMode is configured differently than 
  * config.pinMode. i.e. pinMode(8, INPUT) &  
  * config.pinMode(8, INPUTPULLUP, CHANGE) then you will
  * have to call pinMode(8, INPUT) after waking to have
  * that pin configured back to an INPUT, the library does
  * reconfigure it for you. If both pinMode's are configured
  * the same as in your case you can omit the call to pinMode 
  * after sleep since the library will leave it configured 
  * as such.
  ******************************************************/
}

void loop() {
  static unsigned long ticker;
  if (digitalRead(8) == LOW) {
    if (millis() - ticker > 5000) {
      digitalWrite(13, LOW);
    }

  } else {
    ticker = millis();
    digitalWrite(13, HIGH);
  }
}
```

holgerlembke, 2015-06-09:
Thank you, works!

Perhaps this should be made more clear. I grant that example.... ;-)

closed 2015-06-09 by holgerlembke