duff2013/Snooze #3: dynamic cpu speeds

ramonschepers, 2015-10-23:
i'm missing this feature, and i find it weird why you did remove it.

purely so i can keep running my code without hibernating, deepsleep or whatever.

because it would be needed if one wants to dynamically control their cpu speed to keep running and save power. much like a smartwatch could does.

duff2013, 2015-10-23:
The main problem with this was that the Teensy core does not support this. What I mean is that everything is more or less tied to F_CPU, F_BUS, F_MEM speeds defined at compile time. So for an example, the "delay" function is tied to F_CPU and changing the CPU speed during runtime makes the delayed time wrong (very simplified answer). You have to roll your own delay function for dynamic CPU speeds.

So my thought process was that the majority of users would just compile their CPU a low speeds instead and I wanted to make this library as focused as possible. What happens is feature creep where everyone wants their hardware feature to work with dynamic speeds and then the library gets to bloated.

While saying that, I am still considering using dynamic speeds but just not provide support for the additional hardware tweaks that would be needed to make it play nice at any CPU, BUS or MEM speed. I would leave it to the user to roll their own tweaks?

ramonschepers, 2015-10-23:
well, if you want the snooze lib focused, you can write another one to set cpu speeds (for example: 72mhz,36mhz,18mhz,9mhz,4,5mhz,2,25mhz... so basically like a divide by 2)

and writing own delay functions won't be that hard if one knows what he's doing.

p.s. i wish i had the knowledge to adapt the lowpower_teensy3 lib to get it to work again to save effort on your side

duff2013, 2015-10-23:
It would probably only support the compiled speed and one low speeds (2 MHz?) and by support meaning only getting the processor to that speed and back. All other functionality would have to be implemented by the user.

This pretty much the reason I left the Lowpower_Teensy3 libaray on github but this is directly tied to lowpower operation of the teensy so I'm inclined to add it in just not sure when yet.

ramonschepers, 2015-10-23:
ah ok, i hope it'll come soon again! :)

duff2013, 2015-10-27:
I added it, check out the new example and let me know what you think?

ramonschepers, 2015-10-27:
well, not exactly what i am looking for, because it reruns the loop at stock speed instead of 2mhz again, i prefer to have something like calling a void Set_CPUspeed(uint32_t speed) switches cpu speeds.

duff2013, 2015-10-27:
Well, that is true but it gets confusing to know what speed you're running at that way. This way you know for sure. You can just put while loop inside of it if you need to stay @ 2MHz and "break" outof it when needed.

ramonschepers, 2015-10-27:
true that, but it won't be usefull as i need to write a loop function for each speed? or am i wrong?

duff2013, 2015-10-27:
Either way you have to separate your code for the two CPU speeds, this makes it more clear when you are running @ 2 MHz. 

You said you're making a watch, well I assume your running at 2 MHz when the user is not doing much so you could do something like this. 
```c
void loop() {
  "do stuff at normal speed....."

  //-----------------------------------------------
  if ("Go into 2 MHz mode") {
    REDUCED_CPU_BLOCK() {
      while ("if this is true, run @ 2 MHz") {
        "...do limited stuff @ 2 MHz"
      }
    }
  }
  //-----------------------------------------------

  "do stuff at normal speed....."
}
```
see how it's encapsulated and now you can rewrite all the core functionalities to use at 2 MHz in the 'REDUCED_CPU_BLOCK' and not get confused when you are actually running at 2 MHz instead of normal CPU speed.

The other way it I have add a bunch of supporting functions for the user to let them know what CPU speed the code is currently running which makes the Snooze class more cluttered, this way didn't add any to Snooze class. I know it doesn't look the same but it works the same with what I see as a benefit of easily knowing when you are running at what CPU speed.

ramonschepers, 2015-10-27:
well, my suggestion would be to let the function return the speed in mhz (like for example 24 for 24mhz) so the user can check in their own code what they wanna do with it.

like 

uint8_t speed;

void setup() {
  speed = SET_CPU_SPEED(24); //returns the value if succes, 
  //returns 0 and load default speed if error
  if (speed == 0) {
    //print out error on serial with wrong cpu speed!
    speed = F_CPU;
  }
}
void loop() {
  delay2(1000); //delay for a second
}
void delay2(uint8_t millis) {
  if (speed == 24) {
    //do timings/delays related to 24mhz
  }
}

duff2013, 2017-01-04:
I'm going to close this issue.

closed 2017-01-04 by duff2013