duff2013/Snooze #5: RTC not advancing in deepSleep

cunidev, 2016-01-03:
I don't know what can be the cause, or whether this is voluntary and/or needed in order for this library to run, but I notice that the Teensy RTC stops when I put it in deepSleep... And this can cause problems to many applications that need the RTC to run in background. I'm using a Teensy 3.1 board. Thanks in advance.

duff2013, 2016-01-03:
Assuming you have the crystal installed and your using the stock time library that comes with teensyduino, you need resync (setSyncProvider) the RTC after waking up. Not sure what you mean by not working? I know it works for me after waking but again I have to resync it so the time library is happy.

cunidev, 2016-01-05:
Thanks, I didn't know I had to resync. What about implementing that by
default in the library, if it's necessary?

duff2013, 2016-01-05:
I'm currently updating the library anyway so I'll look into that.

duff2013, 2017-01-04:
I'm going to close this issue.

closed 2017-01-04 by duff2013