duff2013/Snooze #7: Current wakeup from deepSleep takes ~350us, can it be sped up?

goverd, 2016-03-12:
First off thanks for this great library!
I've been using it extensively over the last year, but what I noticed is that the time from pin interrupt occurring to code getting back to the 'main' section is around 350 us. From what I can see the majority of the tasks that are bing accomplished after the wakeup are fairly straightforward and should complete in a fraction of that time, the only reason for a such a long delay seems to be the transition from PBE to PEE, could you confirm? Also, any suggestions on how to speed it up?

duff2013, 2016-03-12:
I'll take a look at this, one thing is that all drivers are dealt with even if you don't use them so this could be the issue, meaning low power timer, compare, rtc, etc drivers get called. Though you might be right and the wakeup isr that clears the flags of the drivers and transitions from pbe->pee could be also causing that delay. I have a new version in the works that only handles the ones you use before and after wakeup. Also what sleep mode are you using 'deepSleep' or 'hibernate'?

goverd, 2016-03-14:
Great, thanks for looking into this. I'm using deepSleep, since need the regulator to stay on to provide 3V to other modules and I'm using Teensy 3.1 at the moment...

duff2013, 2017-04-15:
This will be closed in day or two unless you still have this problem.

closed 2017-08-19 by duff2013