duff2013/Snooze #8: USB-HID device failure after call to 'sleep' or 'deepSleep'

i-e-b, 2016-04-14:
Hi.

I have a Teensy 3.0 set up as a USB-HID device acting as a keyboard.
Arduino IDE 1.6.8, Teensyduino 1.28. Running at 48MHz.

My `setup` contains
```cpp
  config.setTimer(35);
  config.setPeripheral = USB_ON;
  Keyboard.begin();
. . .
```

And my `loop` is like
```cpp
. . .
  rescanKeyboard();
  
  // Snooze.idle();  // this works, but has a minimal effect
  // Snooze.sleep(config); // this pauses, but breaks USB-HID
  // Snooze.deepSleep(config);  // this pauses, but breaks USB-HID
  delay(35); // the old, power hungry code
. . . // do keyboard logic . . .
  digitalWrite(ledPin, (keyDown) ? HIGH : LOW);
```

If I call `Snooze.idle`, the USB device continues to work, but the delay is very small (I haven't checked with a scope, but it's around 500us)
The `sleep` and `deepSleep` calls pause and resume effectively -- I can see a big power consumption drop, and the 'key-down' led lights as expected.
However, the USB HID device no longer works (does not show up as a device) as soon as the delay is called.
If I reduce `config.setTimer` to 1 or 0, the USB still dies.

This failure also happens if I use the `REDUCED_CPU_BLOCK` call at any point.

Power consumptions are (approx):
```
plain delay()            -- 130mw
for() loop around idle() -- 127mw
sleep() / deepSleep()    -- 50mw
```

Can you help?

Many thanks.

duff2013, 2016-04-15:
I've been traveling as of late I'll be back next week.

duff2013, 2016-04-23:
How do i use -> rescanKeyboard();

USB is disabled during any sleep so you might need a longer delay after waking like 700ms. This is a limitation of sleeping the teensy and using USB. But to be sure it could work i need a simple test case sketch that shows the issue and can compile.

i-e-b, 2016-04-24:
Thanks for your time on this.

If the USB is disabled, that might be a show stopper -- HID specifies a device as 'faulty' if fails to respond to a poll every 500ms, and I think most OSs are stricter than that.

I found that `for (i=0;i < 50; i++){ Snooze.idle();}` gave a delay without killing the USB connection, but `millis()` counting seems to be broken when I did that (I guess the non-RTC timer gets sleeped?) and so getting a reliable delay seems tricky.

A minimal sketch:
```cpp
#include <Snooze.h>
SnoozeBlock config;

void setup() {
  config.setTimer(100);
  config.setPeripheral = USB_ON;
  Keyboard.begin();
  delay(1000); // Windows can't accept input immediately after a keyboard is attached
}

int x = 0;
void loop() {
  Snooze.sleep(config); // comment this out
  //delay(100);         // and this in to test without sleep

  if (x++ == 10) {
    Keyboard.print("still working");
  }
}
```

That should type "still working" after 2 seconds.

For interest, here is the code for the project (please excuse my sloppy code) : https://gist.github.com/i-e-b/2ae28cd4e46a3113ea04/343ff0c9ceca36cbcd7918e41d587074f70219a8
`setup` is on line 162, `loop` is on 205.

duff2013, 2016-04-25:
The clocks are stopped in all sleep modes so it takes time to reconnect with the pc after waking up from sleep. You need to let the device reconnect before trying to print to the keyboard. Try a delay of 700ms after waking before using the keyboard.

 I'm still developing a new and improved version of Snooze that I hope to release by the end of the week that will address the idle problem and let the USB stay connected during "sleep".

duff2013, 2017-01-04:
I made USB Serial driver that might help but I'm going to close this issue, version 6 is released.

closed 2017-01-04 by duff2013