duff2013/Snooze #11: Can we configure wakeup for 3 seconds longhold of button press?

mituldesai, 2016-06-14:
Hi,
Thanks a lot for this amazing library. We are using your library to put our teensy3.2 enabled prototype into sleep. I am configuring 2 pin of teensy as input pull up to wake up teensy. I have connected a active low detection switch with 2nd pin. Hence, Teensy is waking up fantastically on pressing switch after sleep.
According to utility of our product, Teensy should wake up from sleep after button press of 3 seconds hold. Now, as far as I understand your library, if i press button & release (irrespective of for how much seconds it pressed), teensy takes it as a single press & wakes up. So when we configure 2nd pin of teensy as wake up pin in void setup(), is there any instruction set by which I can configure wake up detection after 3 seconds of long hold?. So that Teensy only wakes up after 3 seconds of button press or in other words, teensy considers 3 seconds long hold button press as wake up signal.
If you can give us a solution or any config settings, your library can satisfies our this functionality also.
Thanks again for awesome library.

duff2013, 2016-06-14:
Not really, could you configure the pin as (INPUT_PULLUP) and wakeup on FALLING edge, then check how long the button is held for, if its not held for 3 seconds go back to deep sleep? 

That Kinetis processer has no wakeup ability for this specifically.

FrankBoesing, 2016-06-14:
Well..how much energy is wasted during <3 seconds ?
just make  it sleep again if the button-press was not long enough,,,

duff2013, 2016-06-14:
Actually, I've seen consumer products that have this exact functionality. My baby monitor works like this when waking it up. I have to hold a button for a couple of seconds and the same when powering it down. 

I was thinking about this and maybe my [Snooze V6](https://github.com/duff2013/Snooze_V6_Beta) could have a driver that does this. I was thinking that you could wakeup in VLPR mode @2MHz cpu speed which would be 3 seconds or whatever at ~1mA instead of full power.

FrankBoesing, 2016-06-14:
dunno if it is worth the work..those few seconds make no real difference 
(well, max 3 seconds less if powered with battery) - but it's your child 
:) (any you can be proud of it :)

duff2013, 2016-06-14:
I agree the amount of power savings is probably minimal for only 3 seconds.

mituldesai, 2016-07-04:
I took all inputs from you guys. I built a code based on following flow:
1. teensy goes into sleep
2. On button press it wakes up.
3. checks if pressed for 3 seconds long hold.
4 If not goes back into sleep.
5. If yes goes to wake up function.

Program works perfectly, If I dont put on sleeping function in 3 seconds checking loop & if just blink LEDs on 3 second long press. But If I put sleep function here, then once it wakes up , then checks for 3 seconds, If not then it goes into continues sleep. Reason is that his library has not ACTIVE LOW detection. It has only FALLING EDGE interrupt. So, Once I press the button, then in 3 seconds checking loop, It cant checks the pressed event, as It only detects the initial FALLING EDGE. I am pasting my code below.  

in following code, If I comment sleeping instructions in loop, initially teensy stays for 6 seconds turning a red LED on. Then it goes into sleep by turning red LED off. After pressing button for 3 seconds, green LED stays on for 3 seconds then Teensy come out of 3 secods loop & blinks 13 pin LED.

On uncommenting  sleeping instructions. on pressing button for wakeup, green LED stays on only, which means teensy goes into permenant sleep because of lack of LOW ACTIVE detection

#include <Snooze.h>

SnoozeBlock config;
long sleepDelay = 2000;
long lastSleepTime;   
bool slip = false;
int lastState = HIGH;
int reading;

void setup() {
  // put your setup code here, to run once:
   pinMode(13, OUTPUT);
   pinMode(4, OUTPUT);
   pinMode(6, OUTPUT);
   config.pinMode(2, INPUT_PULLUP, LOW);//pin, mode, type
//   config.pinMode(2, CMP, LOW, 0.03);//pin, mode, type, threshold(v)
}

void loop() {
  // put your main code here, to run repeatedly:
    
digitalWrite(4, HIGH);
delay (6000);
digitalWrite(4, LOW);
 int who = Snooze.hibernate( config );// return module that woke processor

  if (who == 2) { // pin wakeup source is its pin value
    
     reading = digitalRead(2);
      

     lastSleepTime = millis();

      
     if (who == 2)
     {
      while (millis() - lastSleepTime <= sleepDelay)
      {    
       digitalWrite(6, HIGH);
       config.pinMode(2, INPUT_PULLUP, LOW);
       who = Snooze.hibernate( config );
      }
      digitalWrite(6, LOW);
     }
      for (int i = 0; i < 6; i++) {
        Serial.println(lastSleepTime);
            digitalWrite(13, HIGH);
            delay(400);
            digitalWrite(13, LOW);
            delay(400);
        }
}
}

duff2013, 2016-07-05:
Would you be willing to upgrade to [version 6 of snooze](https://github.com/duff2013/Snooze_V6_Beta)? Its where I'm going with this library and has many upgrades and fixes. I have example below of a 3 second button hold before waking already written up, you just need snooze v6:) 

The new Snooze has few changes in how things get initialize but works pretty much the same way as version 5.  The code below you can see the difference in how you do things between the two versions but the sleeping part is exactly the same.

```c
/*
    This example is used when you want to simulate a
    button being held for certian amount of time
    before waking your Teensy from hibernate sleep.

    This example uses the Bounce library to make sure
    the button is still pressed for 3 seconds after
    waking up from low power mode. If it released
    before the 3 seconds are up, go back to sleep.
*/

#include <Snooze.h>
#include <Bounce.h>

// Load drivers
SnoozeDigital digital;// this is the pin wakeup driver

// use bounce for pin 2, debounce of 5ms
Bounce button = Bounce(2, 5);

// install driver into SnoozeBlock
SnoozeBlock config(digital);// digital driver is now a wakeup

long sleepDelay = 2000;
long lastSleepTime;
bool slip = false;
int lastState = HIGH;
int reading;

void setup() {
  // Configure pin 2 for bounce library
  pinMode(2, INPUT_PULLUP);
  // debug led
  pinMode(LED_BUILTIN, OUTPUT);
  while (!Serial);
  delay(100);
  Serial.println("start...");
  delay(20);
  //pin, mode, type
  digital.pinMode(2, INPUT_PULLUP, FALLING);
}

void loop() {
  // if not held for 3 sec go back here to sleep.
SLEEP:
  // you need to update before sleeping.
  button.update();
  // returns module that woke processor after waking from low power mode.
  Snooze.hibernate( config );
  // indicate the button woke it up, hold high for as long as the button
  // is held down.
  digitalWrite(LED_BUILTIN, HIGH);
  elapsedMillis timeout = 0;
  // bounce needs to call update longer than the debounce time = 5ms,
  // which is set in constructor.
  while (timeout < 6) button.update();
  // check for
  bool awake = threeSecondHold();
  if (!awake) goto SLEEP;

  // do some stuff for 7 seconds then go to sleep.
  elapsedMillis time = 0;
  while (1) {
    
    unsigned int t = time;
    Serial.printf("doin stuff for: %i milliseconds\n", t);
    
    // back to sleep after 7 seconds
    if (time > 7000) {
      Serial.println("sleeping now :)");
      // little delay so serial can finish sending
      delay(5);
      goto SLEEP;
    }

    digitalWrite(LED_BUILTIN, HIGH);
    delay(50);
    digitalWrite(LED_BUILTIN, LOW);
    delay(50);
  }
}

bool threeSecondHold() {
  reading = button.duration();
  //delay(1);
  // pin wakeup source is its pin value
  // this is the 3 sec button press check
  while (button.duration() < 3000) {
    // get the current pin state, must have this!
    button.update();
    // check the pin 2 state, if button not
    // pressed before 3 seconds go back to
    // sleep.
    if (button.read() != 0) {
      digitalWrite(LED_BUILTIN, LOW);
      // let go of button before 3 sec up
      return false;
    }
  }
  digitalWrite(LED_BUILTIN, LOW);
  // button was held for 3 seconds so now we are awake
  return true;
}
```

mituldesai, 2016-07-06:
Hi, thanks a lot for this new library. we really appreciate yr support. I downloaded version 6 library from: https://github.com/duff2013/Snooze_V6_Beta

I extracted the zip files & copied the Snooze folder into arduino libraries. I removed old Snooze library. I tried deepsleep_all_wakeups example, but I got the following errors:

Arduino: 1.6.7 (Windows 10), TD: 1.27, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz optimized (overclock), US English"

In file included from C:\Users\HP\Documents\Arduino\libraries\Snooze/SnoozeBlock.h:36:0,
from C:\Users\HP\Documents\Arduino\libraries\Snooze/Snooze.h:35,
from C:\Users\HP\Documents\Arduino\libraries\Snooze\examples\deepsleep\deepSleep_all_wakeups\deepSleep_all_wakeups.ino:6:
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h: In function 'void llwu_configure_pin_mask(uint8_t, uint8_t)':
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:124:73: error: 'LLWU_PE4_WUPE12' was not declared in this scope
if ( pin == 2 )       mask->PE4 |= LLWU_PE4_WUPE12( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:125:73: error: 'LLWU_PE2_WUPE4' was not declared in this scope
else if ( pin == 4 )  mask->PE2 |= LLWU_PE2_WUPE4 ( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:126:73: error: 'LLWU_PE4_WUPE14' was not declared in this scope
else if ( pin == 6 )  mask->PE4 |= LLWU_PE4_WUPE14( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:127:73: error: 'LLWU_PE4_WUPE13' was not declared in this scope
else if ( pin == 7 )  mask->PE4 |= LLWU_PE4_WUPE13( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:128:73: error: 'LLWU_PE2_WUPE7' was not declared in this scope
else if ( pin == 9 )  mask->PE2 |= LLWU_PE2_WUPE7 ( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:129:73: error: 'LLWU_PE3_WUPE8' was not declared in this scope
else if ( pin == 10 ) mask->PE3 |= LLWU_PE3_WUPE8 ( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:130:73: error: 'LLWU_PE3_WUPE10' was not declared in this scope
else if ( pin == 11 ) mask->PE3 |= LLWU_PE3_WUPE10( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:131:73: error: 'LLWU_PE3_WUPE9' was not declared in this scope
else if ( pin == 13 ) mask->PE3 |= LLWU_PE3_WUPE9 ( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:132:73: error: 'LLWU_PE2_WUPE5' was not declared in this scope
else if ( pin == 16 ) mask->PE2 |= LLWU_PE2_WUPE5 ( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:133:73: error: 'LLWU_PE4_WUPE15' was not declared in this scope
else if ( pin == 21 ) mask->PE4 |= LLWU_PE4_WUPE15( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:134:73: error: 'LLWU_PE2_WUPE6' was not declared in this scope
else if ( pin == 22 ) mask->PE2 |= LLWU_PE2_WUPE6 ( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:135:73: error: 'LLWU_PE1_WUPE0' was not declared in this scope
else if ( pin == 26 ) mask->PE1 |= LLWU_PE1_WUPE0 ( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:136:73: error: 'LLWU_PE3_WUPE11' was not declared in this scope
else if ( pin == 30 ) mask->PE3 |= LLWU_PE3_WUPE11( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:137:73: error: 'LLWU_PE1_WUPE3' was not declared in this scope
else if ( pin == 33 ) mask->PE1 |= LLWU_PE1_WUPE3 ( detect_type );
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h: In function 'void llwu_configure_modules_mask(uint8_t)':
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:151:56: error: 'LLWU_ME_WUME0' was not declared in this scope
if( module & LLWU_LPTMR_MOD )      mask->ME |= LLWU_ME_WUME0;
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:153:56: error: 'LLWU_ME_WUME5' was not declared in this scope
else if ( module & LLWU_RTCA_MOD ) mask->ME |= LLWU_ME_WUME5;
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:154:56: error: 'LLWU_ME_WUME7' was not declared in this scope
else if ( module & LLWU_RTCS_MOD ) mask->ME |= LLWU_ME_WUME7;
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:156:56: error: 'LLWU_ME_WUME4' was not declared in this scope
else if ( module & LLWU_TSI_MOD )  mask->ME |= LLWU_ME_WUME4;
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:157:56: error: 'LLWU_ME_WUME1' was not declared in this scope
else if ( module & LLWU_CMP0_MOD ) mask->ME |= LLWU_ME_WUME1;
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/wake.h:158:56: error: 'LLWU_ME_WUME2' was not declared in this scope
else if ( module & LLWU_CMP1_MOD ) mask->ME |= LLWU_ME_WUME2;
[... 58 lines cut ...]
^
In file included from C:\Users\HP\Documents\Arduino\libraries\Snooze/Snooze.h:38:0,
from C:\Users\HP\Documents\Arduino\libraries\Snooze\examples\deepsleep\deepSleep_all_wakeups\deepSleep_all_wakeups.ino:6:
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/SnoozeTouch.h: In constructor 'SnoozeTouch::SnoozeTouch()':
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/SnoozeTouch.h:73:5: error: expected identifier before '{' token
{
^
In file included from C:\Users\HP\Documents\Arduino\libraries\Snooze/Snooze.h:39:0,
from C:\Users\HP\Documents\Arduino\libraries\Snooze\examples\deepsleep\deepSleep_all_wakeups\deepSleep_all_wakeups.ino:6:
[... cut ...]
uint8_t CR0;
^
C:\Users\HP\Documents\Arduino\libraries\Snooze/utility/SnoozeCompare.h:62:5: warning:   when initialized here [-Wreorder]
SnoozeCompare(void)  :
^
C:\Users\HP\Documents\Arduino\libraries\Snooze\examples\deepsleep\deepSleep_all_wakeups\deepSleep_all_wakeups.ino: In function 'void setup()':
deepSleep_all_wakeups:80: error: 'class SnoozeTouch' has no member named 'pinMode'
touch.pinMode(0, touchRead(0) + 250); // pin, threshold
^
exit status 1
'class SnoozeTouch' has no member named 'pinMode'

duff2013, 2016-07-06:
Sorry about that, you need latest teensyduino core from github, give me a few hours and I will upload a file that will make it compile until Teensy 1.29 is released.

duff2013, 2016-07-06:
Can you install the latest [Teensyduino beta4](https://forum.pjrc.com/threads/35315-Teensyduino-1-29-Beta-4-Available)?

mituldesai, 2016-07-07:
Okay sure, I will download it & will try examples again. But, why its not working with current Teensydiuno, previous library works fine.

duff2013, 2016-07-07:
Because of the new Teensy 3.5 support.

mituldesai, 2016-07-07:
Okay, So the link you have provided is latest with support? I have downloadded it. I am installing right now.

mituldesai, 2016-07-07:
Hii, after installing , I have check your button hold example code. Its working awosomely. you have done an incredible job. I cant explain you how much its important for us.  When we teensy users stuck with lack of support for arduino standard sleep libraries, you have compe up with a very good library. When I report you an issue, you provided a whole example code. Thats shows your conviction. I would like to congratulate you again & thanks a lot for all your support. I would love to seeing more upgrades from you.

duff2013, 2016-07-07:
Thanks, I have added your issue as an example sketch also! I still have some ideas for expanding on it by using the REDUCED_CPU_BLOCK while doing the button on check. This way your not running at full power to just check if a button is pressed.

mituldesai, 2016-07-07:
Yeah, thats what, you added my issue as example sketch , that was true conviction of yours. can you ping me your updated library here on this space? Once you complete the above mentioned functionality.

duff2013, 2017-01-04:
I'm going to close this issue.

closed 2017-01-04 by duff2013