duff2013/Snooze #12: DAC not switched off in hibernate

jpmeijers, 2016-07-11:
A full description of my problem can be seen at [https://forum.pjrc.com/threads/35323-Teensy-LC-DAC-on-during-hibernate](https://forum.pjrc.com/threads/35323-Teensy-LC-DAC-on-during-hibernate).

Maybe this was done on purpose so that the voltage on the DAC pin do not change when going into sleep mode. But it will still be useful in some situations to switch off the DAC and save 200µA.

I am using the Snooze library version: 1/12/16 v5.0.1. Any newer ones I could not get compiling for the Teensy LC.

duff2013, 2016-07-11:
I'll take a look at how to turn off the DAC before sleeping for you. 

As for the LC issue, let me look into it but really I'm only working on [Version 6](https://github.com/duff2013/Snooze_V6_Beta) now so if there is problem with the Teensy LC and [Version 6](https://github.com/duff2013/Snooze_V6_Beta) i'll fix that version not the older version 5.

duff2013, 2016-07-11:
Well I can't find my Teensy LC in my pile of ever collecting electronics. I hope I can find it soon, but Sooze V6 does not compile yet I will have a working version I hope by tonight if not tomorrow. The only problem is that I can't test it out.

jpmeijers, 2016-07-12:
I successfully turned off the DAC and saved >200uA with the following two lines of code:
DAC0_C0 = ~DAC_C0_DACEN; //DAC_C0_LPEN
SIM_SCGC6 |= SIM_SCGC6_DAC0;// enable DAC0 clock - no influence

Switching off the DAC workied perfectly (line 1), while switching off the clock (line 2) does not seem to have any effect. The order of the two lines also did not have an influence as far as I could see.

I'd love to be able to use V6 on my Teensy LC, so I'll keep an eye on the commits. Thanks for the excellent support!

duff2013, 2016-07-12:
Great, that looks right to me, [Version 6](https://github.com/duff2013/Snooze_V6_Beta) should be ready to go. I found a Teensy LC to test it so go ahead and try it out. A few API changes where made but they you should be able to adapt to the new api fairly easy. All the examples will show you how.

closed 2017-04-15 by duff2013