duff2013/Snooze #13: config error for Teensy LC?

LenShustek, 2016-10-15:
Compiling  pushbutton_pullup_hibernate.ino  example (and my application too) generates this error:
 
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\Snooze.cpp:210:33: error: 'rtc_alarm_set' was not declared in this scope
     rtc_alarm_set( &p->rtc_mask );

duff2013, 2016-10-16:
can you try[ version 6](https://github.com/duff2013/Snooze_V6_Beta)?

LenShustek, 2016-10-16:
Ok, I'll put that on my to-do list. It's a bit of a project because I then need to set up the [microammeter ](https://github.com/LenShustek/microammeter) to verify the the power utilization, but I'll get to it.

duff2013, 2016-10-16:
I'm going to merge V6 to the main Snooze repository sometime in the near future.

MarcianoPreciado, 2016-11-19:
Could you please look into this again? I'm really pressed for time on a project and can't seem to get any version of snooze to work with my Teensy LC sketch.

LenShustek, 2016-11-19:
I too was in deep trouble for a production project that used Snooze. Version 5 stopped working with Teensy LC, and version 6 doesn't work yet either; see the isssue at https://github.com/duff2013/Snooze_V6_Beta/issues/5#issuecomment-256461806

I was up the creek but luckily have a paddle: I kept using an old version (5.5.1?) that still works.  If you can't extract that from the repository, let me know and I'll post it somewhere.

I guess Colln has been busy with other stuff, which I can understand, but he's left his fans in a difficult spot for months.

duff2013, 2016-11-19:
It just checked, this version 6.2.3 works with the LC. Make sure you delete all copies of Snooze before installing it?

LenShustek, 2016-11-20:
It does indeed. Thanks!

closed 2016-11-20 by LenShustek