duff2013/Snooze #14: power consumption comparison between Teensy3.1 and TeensyLC

CongducPham, 2016-11-15:
Hi, first of all, thank you for your great work!

I use your library on a Tennsy32 (no modification) and measure hibernate power consumption of about 43uA. Did you manage to have less than that?

I'm also wondering whether using an TeensyLC would provide further lower consumption or not. I do not have the TeensyLC and if you measure lower power consumption, then it will definitely be interesting for me to buy some.  Thank you for any answer you can provide.

regards,

duff2013, 2016-11-15:
I saw around 15uA with nothing attached to Teensy 3.2 and around 5uA on a TeenyLC with nothing attached. I actually did redundant measurements with three different types of current detecting equipment.

CongducPham, 2016-11-15:
Thank for the answer. I also have nothing attached to the Teensy32 and the lowest in hibernate is the 43uA. I have version 5.5, would version 6 reduce power consumption? So seems that the LC does reduce the power consumption. I'm going to get some of them to test. Thanks again for your excellent work!

duff2013, 2016-11-15:
Version 6 would not matter since they both share pretty much the same driver code for the actual low power operation. That being said I'm no longer going to develop version 5 anymore and will merge version 6 in the near future. Works almost the same as version 5, version 6 makes it easier for me to add new features.

Back to the actual measurements, can tell me how you are measuring for current (equipment). Make sure nothing is attached to the teensy and can you share the code you are using?

CongducPham, 2016-11-16:
I'm using the a variant of the example for hibernate:

#include <Snooze.h>
SnoozeBlock config;
void setup() {
    Serial.begin(38400);
    pinMode(LED_BUILTIN, OUTPUT);
    config.setTimer(30000);// milliseconds
}

void loop() {

    Serial.println("No low power for 10s");
    Serial.flush();
    delay(10000);

    Serial.println("Entering low power for 30s");
    Serial.flush();
    delay(100);        
    /********************************************************
     * feed the sleep function its wakeup parameters. Then go 
     * to deepSleep.
     ********************************************************/
    //int who = Snooze.deepSleep( config );// return module that woke processor
    int who = Snooze.hibernate( config );

    Serial.print("Wake up from low power with code: ");
    Serial.println(who);
    Serial.flush();
}

Nothing is attached. To measure consumption I use traditional 1ohm schematic to measure voltage between the 2 side of the resistor. I also used the meter in amp mode tuned to uA and the measures are consistent. I'm injecting 5v to Vin. When in hibernate mode, drawn current is about 43uA. When active, the drawn current is about 32mA.

duff2013, 2016-11-16:
Hmmm, try taking all the print statements out. USB doesn't work well at all when using sleep, not sure if that will do anything for the current but in Hibernate the USB regulator is put in low power state so maybe if anything is in the usb queue it could effect the regulator. I measured and other have confirmed the ~15uA for the Teensy3.2. Have you tried your measurement on other Teensy 3.2's? I also used a shunt resistor a multimeter and specialized low current measuring tool that all agreed on multiple Teensy's. Not saying your measurements are wrong I just not sure why it is giving you these readings.

CongducPham, 2016-11-16:
OK, I will try to remove the print statement, but I assumed that the 
flush() would actually flush the queue. It the print statements do 
reduce the efficacy of the low power, then in production mode, I have to 
remove all print statements. I don't have the specialized tool, but the 
difference is quite high currently, so will try first what you say and 
will get back to you. Thanks for the answer.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

duff2013, 2016-11-16:
flush for USB (Serial) does not do anything. https://github.com/PaulStoffregen/cores/issues/43

CongducPham, 2016-11-17:
> flush for USB (Serial) does not do anything. PaulStoffregen/cores#43

OK, I removed all print statements and it does not change anything, 
still about 43uA in hibernate.

I took another Teensy32, this one shows 50uA.

It is far from the 15uA that you manage to get. Could you share the code 
with which you measure the 15uA?

Regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

duff2013, 2016-11-17:
just what you posted is good, maybe the new regulator on the Teensy3.2 Vin power could be accounting for this? I really don't know what else to tell you. I'll get out my equipment next week and see what I come up with since it's been awhile and I need to do some work with the Teensy 3.5/6 and Snooze anyway.

CongducPham, 2016-12-08:
Hi, I confirm that with the TeensyLC 5uA with nothing attached can be achieved. But for the Teensy32, still it is around 40uA.

regards,

duff2013, 2016-12-08:
The values I reported are with a Teensy 3.1 which used a smaller volatge regulator compared to the 3.2. Your values for the Teensy 3.2 are what I got. You can read about the differences between the 3.1 and 3.2 [here](https://forum.pjrc.com/threads/29774-Teensy-3-2) and [here](https://www.pjrc.com/teensy/teensy31.html). Thanks for following up!

CongducPham, 2016-12-09:
OK, thank you. Then I will also buy some 3.1 has I probably don't need 500mA on the power pin.

duff2013, 2017-01-04:
I'm going to close this issue.

closed 2017-01-04 by duff2013