duff2013/Snooze #15: Digital Interrupts not Working for Teensy LC on Sleep

MarcianoPreciado, 2016-11-25:
When trying out the sleep_all_wakeups example code on my Teensy LC the digital interrupts aren't waking the processor from sleep. I switch the 
`#elif defined(__MKL26Z64__)
    who = Snooze.sleep( config_teensyLC );// return module that woke processor
#endif
`
part for.
`#elif defined(__MKL26Z64__)
    who = Snooze.hibernate( config_teensyLC );// return module that woke processor
#endif`

and the digital interrupt functionality works just fine.

duff2013, 2016-11-26:
Do you have the latest version of Snooze? I have tested the digital wake-ups before for you on the LC and they worked fine for all sleep types i.e.. sleep, deepSleep and hibernate. I'm not sure why yours is not working, what other code and hardware do you have hooked up to the LC? I need to know the details to trouble shoot this.

duff2013, 2016-11-27:
F.Y.I you got me looking for any bugs in the digital pin driver code and I did find one small one that might effect the LC. While I did test that driver with my LC and it did work I didn't try with all pins so maybe there could be certain pins effected with this bug. I don't have an LC with me now but I went ahead and updated this library.

MarcianoPreciado, 2016-12-06:
I ran the example sketch and nothing more. Yes, it was the most recent version, and on a bare-bones teensy LC straight out of the packaging.

duff2013, 2016-12-06:
I don't know, digital wakeups for sleep work just fine with the LC's I have. You ae pulling the wakeup pins to ground to wake right? The example uses Pull-Ups.

closed 2017-02-03 by MarcianoPreciado

darianbjohnson, 2020-01-02:
@MarcianoPreciado did you ever figure out why this wasn't working for you? I'm having the same problem with my "straight out of the bag" Teensy LC

MarcianoPreciado, 2020-01-12:
I believe my issue was that I wasn't configuring my pins correctly. They needed to be INPUT_PULLDOWN

MarcianoPreciado, 2020-01-12:
I'll check my old code to check.

MarcianoPreciado, 2020-01-12:
```_digital.pinMode(_ackBtn, INPUT_PULLUP, RISING);```