duff2013/Snooze #16: ADC, deepSleep, and the missing ~250uA

salvagione, 2016-12-31:
First, great library! I've used it on a few projects and recommended it often. I'm moving a project over to v6 and going for absolute minimum power, which I haven't done in the past. 

I've got a bone stock 3.2 running your button_hold_wakeup example. I'm seeing ~40mA while running and ~270uA while in deepSleep. I'm using David Jones' uCurrrent gold to measure the current.

A google search hints at turning off the ADC. Is that done in your library or do I have to diddle the bit elsewhere?

duff2013, 2016-12-31:
I have uCurrent Gold also! No none of the drivers will configure the analog pins to anything else, so you would have to do that. The digital wake-up driver will configure them for digital pin and back to whatever they where before hand but as far your question the answer is you will have to take-care of that. 

The current you are seeing is because the Teensy 3.2 uses a beefier regulator than the 3.1.

salvagione, 2016-12-31:
Yes, the uCurrent Gold has been really nice in helping me get to the bottom of power usage.
So, can I make a request that shutting down the ADC would be an amazing feature of the Snooze library? Or is the majority of the ~250uA 'overage' the voltage regulator?

salvagione, 2016-12-31:
Hmm, was this the right direction?
https://forum.pjrc.com/threads/33028-Teensy-3-1-Save-Power-Disable-ADC
The link you reference is dead.

duff2013, 2016-12-31:
Actually disabling the ADC had very minimal effect on processor power usage from what I saw. Maybe you can check it and if you can show me that it does I'll consider adding another driver for that.

Yes the majority of the current you are seeing is from that regulator if you have nothing else hooked up to the Teensy.

salvagione, 2017-01-03:
You are correct shutting down the ADC makes little if any difference. Please close this issue and thank you for your time.

salvagione, 2017-01-03:
You are correct shutting down the ADC makes little if any difference. Closing this issue.

closed 2017-01-03 by salvagione