duff2013/Snooze #18: // Using Serial.end() to plug a power leak on an un-powered serial port //

salvagione, 2017-01-03:
It appears Snooze doesn't _shut down_ the serial ports before _hibernating_. I'm using serial2 and serial3 on the Teensy 3.2 and I'm seeing ~13mA draw for each port in the _hibernate_ mode. I'm guessing that the devices on these ports are pulling power even though they are powered down. If I use Serial.end() on both ports before calling Snooze the _power leak_ is gone. Of course, I need to remember to call Serial.begin() for both ports after I wake back up.

duff2013, 2017-01-03:
Yes, pins configuration matters a lot in low power, SPI also will do this also.

If you look at the examples there is one for a 'skeleton driver class' where you can make your own drivers which could help for your issue with the serial ports. Once you have this class (driver) you can use it like any other driver that is currently part of the core (timer, clock, digital etc..). 

You don't need to use this driver and just call the code in the sketch but it makes it cleaner and gives you more control of what order you want your drivers called.

salvagione, 2017-01-05:
Perfect, I'll work on a driver.

closed 2017-01-05 by salvagione