duff2013/Snooze #19: LC not staying in Hibernate

MarcianoPreciado, 2017-01-24:
My Teensy LC project has been having lifespan issues when running on battery. When I throw it into hibernate mode, the current drops to 2 uA. Not long after initiating hibernation, the current returns to about 5 mA but stays in hibernation mode. I'm not sure what is going on here, but this reduces the battery life by several orders of magnitude. It seems to me that maybe internally most of the aspects of hibernation that effect power draw are being disabled, but the processor is still sleeping and hibernation is nominally active, but not truly.

MarcianoPreciado, 2017-01-24:
I'm going to test it a little more tomorrow and provide more details, but this is all I have for now.

duff2013, 2017-01-24:
Ok, can give me idea of your hardware setup? Does this happen with nothing attached? How are powering the Teensy and do you have USB connected? Can you try 'deepSleep' also?

MarcianoPreciado, 2017-01-27:
I have my LC connected to an RTC module and OLED. Both of these peripherals are being powered by digital output pins on the LC that it switch to LOW just before entering hibernation. (Otherwise they continue to drain power from the MCU). However sometimes when falling back asleep, instead of dropping from 15 mA to 50 uA, it will only drop to 11 mA and never fall further unless reawakened and told to hibernate again.

MarcianoPreciado, 2017-01-27:
I've monitored it a little more, and I've now noticed that when it's in one of the high-power hibernation events, it initially drops down to 50 uA, and jumps back up to 11 mA after about 750 milliseconds.

duff2013, 2017-01-27:
Does this happen when the RTC and OLED are not attached? Also the Teensy can only source 20mA through 4 digital pins i assume you are using these?

MarcianoPreciado, 2017-01-27:
I have them connected to D1 and D2. I've also noticed that it happens exactly every other time. I just tested with deepSleep and am still experiencing the same problem. Now going to try without periphery attached.

MarcianoPreciado, 2017-01-27:
I removed the OLED and RTC during the hibernation cycle and it dropped from 11 mA to 10.5 mA.

duff2013, 2017-01-27:
Only pin 5,16,17,21 can source up to 20mA the rest are max of 5mA. Might want see what each peripherals maximum current is. Even though it say 20mA in reality that is pushing that pins circuitry pretty hard so 10% rule should be applied to give you some headroom.

duff2013, 2017-01-27:
In my tests I didn't see this at all, I'll test out my LC with nothing attached again.

MarcianoPreciado, 2017-01-27:
I tried mine out without anything attached before going into hibernate, and the problem isn't there. It seems to me that once the processor has entered hibernation, the pins in use no longer are explicitly set to low. I've rigourously gone through my code and have been rebuilding it and can't find how it could possibly be changing the current draw during hibernation. 
As for the current draw, I'm measuring current straight from the power source. Without the OLED and RTC attached the MCU already draws 12 mA, so I don't think that maxing the peripherals' current is the problem.

MarcianoPreciado, 2017-01-27:
I've configured my hibernation to only respond to and INPUT_PULLUP on my btnPin
Here is the pseudo code for my hibernation operation here.

>If its time to sleep
>> set OLED power pin LOW
>> set RTC power pin LOW
>> enter sleepCycle
. . .
void sleepCycle 
> Snooze.hibernate(myConfiguration)
> set OLED power pin HIGH
> set RTC power pin HIGH
> initiate communication with periphery
> get time from RTC
> initiate display on OLED
> return

salvagione, 2017-01-27:
Try switching the pins to high impedance.

duff2013, 2017-01-27:
I'm testing now with latest Snooze and Teensyduino 1.32Beta1 with just a timer and don't see this.

closed 2017-03-15 by MarcianoPreciado