duff2013/Snooze #20: "who" return value always -1

smokedsalmonbagel, 2017-01-26:
Hi - great library, I am successfully sleeping the teensy down to the 100 uA range!  I can't seem to figure out why the deep_sleep_all example didn't exactly work right.  The LED never flashed based on the who integer.  When I print the value of "who" I always get -1.  I am entering deep sleep with a timer AND interrupt:
`SnoozeDigital digital;
SnoozeTimer timer;
SnoozeBlock config_teensy32(digital,timer);
setup()
{
digital.pinMode(16, INPUT_PULLUP, FALLING);
timer.setTimer(100000);
}
void loop() {
    int who=0;
    who = Snooze.deepSleep( config_teensy32 );
   //Serial.print(who) / write always shows -1 
}`

Am I missing something in the config?

duff2013, 2017-01-27:
what version of Snooze are you using and I assume this for the Teensy 3.2?

smokedsalmonbagel, 2017-01-27:
I have v6.2.5 (just downloaded) and a Teensy 3.2 I believe...  It looks like a 3.2 but it has a black reset button - not sure if that matters - the chip says mk20dx256.  I am using Arduino 1.0.6 rather than 1.6 to support some other chips / libraries.

duff2013, 2017-01-30:
I'll take a look at this tomorrow.

smokedsalmonbagel, 2017-01-30:
Ok thanks, let me know if you need any more details or tests.

duff2013, 2017-01-31:
Sorry for the delay, I'm working on some other pressing issues that came in before you, I haven't forgot :)

smokedsalmonbagel, 2017-02-02:
Ok - no problem... I also am noticing that when I use the timer functions with very large values the timer seems to off early.  I am wondering if it has been tested with long values like
```
timer.setTimer(1000000);
```
If I remove the timer from the config there are no issues.
I'm going to try to look into the code tomorrow for the "who" issue, but I am not familiar with the library so not sure how far I will get.

smokedsalmonbagel, 2017-02-02:
Hi - ok so this seems like a really naive fix but it actually works for me.  Line 121 of Snooze.cpp:
```
#if defined(__MK66FX1M0__) || defined(__MKL26Z64__)
    p->source = llwu_disable( );
#else 
    //replace this   llwu_disable( ); 
    //with:
    p->source = llwu_disable( );
#endif
    p->disableDriver( );
```
The entire if defined might be able to go away, not sure what the implications will be for others though.   I can confirm this is giving me different numbers for digital (16) and timer (36).
Now to take a look at the issue with the timer - I will open a  new issue for that.

duff2013, 2017-02-03:
If that works with all Teensy's then get rid of ifdef's, not sure what I was thinking:)

JonathanBaurer, 2017-02-16:
I was having the same issue with a Teensy 3.2 after updating from Version 5. Snooze.deepSleep() always returned -1.  Smokedsalmonbagel's solution worked for me as well.

duff2013, 2017-02-17:
Try the new update 2.6.7 should be fixed

duff2013, 2017-04-15:
I'm going to close this in day or two unless you are still having problems with this issue?

closed 2017-04-15 by duff2013