duff2013/Snooze #21: Any interrupt pin wakes up Teensy from hibernate

Darellon, 2017-01-29:
Hi duff, 
i recently started playing around with your snooze library. I've noticed that when i attach an interrupt to a pin (not digital.pinMode but attachInterrupt) the interrupt will wake up my Teensy 3.6 from hibernation.  Is this intended to be like this? As far as i understood the only pin interrupts that should wake up the Teensy are the ones declared by digital.pinMode but my normal Interrupts seem to wake the Teensy too. 

Here is the code i've used:

#include <Snooze.h>
#define but_pin 23
#define but2_pin 10
boolean flag = true;

SnoozeDigital  digital;
SnoozeBlock config_teensy36(digital);
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digital.pinMode(but2_pin, INPUT_PULLUP, LOW);
  pinMode(but_pin, INPUT_PULLUP);
  attachInterrupt(but_pin, button_int, LOW);
}
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  Snooze.hibernate( config_teensy36 );
  digitalWrite(LED_BUILTIN, LOW);
  delay(5000);
  if (flag == true) {
    for (int i = 0; i < 5; i++) {
      digitalWrite(LED_BUILTIN, HIGH);
      delay(200);
      digitalWrite(LED_BUILTIN, LOW);
      delay(200);
    }
    flag = false;
  }
}
void button_int() {
  if (flag == false) {
    flag = true;
  }
}

Darellon, 2017-01-29:
i'm using (in case it matters):
Arduino IDE v1.8.1
Teensyduino v1.35
Snooze v6.2.5

duff2013, 2017-04-15:
I believe this is fixed with the latest version, I'm going to close this in the next day or two unless you still have and issue?

closed 2017-04-22 by duff2013