duff2013/Snooze #22: LC pin 17

manitou48, 2017-02-11:
v5 of the lib used to manage the LC's pin 17 by setting it to OUTPUT  LOW.  You might wish to add that feature to current lib, or warn the user in the README

discussed at https://forum.pjrc.com/threads/41812-Snooze-library-attachInterrupt-wakes-up-Teensy-from-hibernate?p=132532&viewfull=1#post132532

duff2013, 2017-02-11:
I have this queued up for my next update in the next few days.

manitou48, 2017-02-24:
it doesn't look like LC pin 17 issue was handled in your latest release?
bump.

duff2013, 2017-02-25:
I did but I made driver class for it. You can see [here](https://github.com/duff2013/Snooze/blob/master/examples/deepsleep/deepSleep_all_wakeups/deepSleep_all_wakeups.ino#L15).

closed 2017-02-28 by manitou48