duff2013/Snooze #27: Serial output is corrupted after waking

michaelmargolis, 2017-04-15:
Tested using sleep and hibernate, this code is based in a paired down  sleep_all_wakeups example.  
```
#include <Snooze.h>

SnoozeDigital digital;

SnoozeBlock config_teensy32(digital);

void setup() {
    Serial3.begin(57600); //Tx on pin 8
    Serial3.println("Debug Serial started");
    pinMode(LED_BUILTIN, OUTPUT);
    digital.pinMode(10, INPUT_PULLUP, RISING);//pin, mode, type       
}

void loop() {
    int who;   
    Serial3.println("Sleeping");
    who = Snooze.hibernate( config_teensy32 );// return module that woke processor    
    if (who == 10) { // pin wakeup source is its pin value
        Serial3.println("triggered by pin 10");   
        for (int i = 0; i < 2; i++) {
            digitalWrite(LED_BUILTIN, HIGH);
            delay(200);
            digitalWrite(LED_BUILTIN, LOW);
            delay(200);
        }
    }     
}
```
Here is the output:
```Debug Serial started
Sleeping
triggered by pin 10
ÿ*[YYÿZÿVHÿÿWÿÿVÿÿÿÿ$ÿ^HÿZÿÿ10
ÿÿÿÿÿÿW-ÿVHÿÿWÿÿVÿÿÿÿ$ÿ^HÿZÿÿ10
ÿ
```

The corruption occurs following the first wake up.  subsequent interrupts do flash the LED but the serial output remains corrupted. 
Tested with a teensy3.1 and Snooze version 6.2.9 using Arduino IDE 1.8.1

duff2013, 2017-04-15:
See the [hibernate_usb_serial](https://github.com/duff2013/Snooze/tree/master/examples/hibernate/hibernate_usb_serial) example for a possible solution. I'm going to try to see if Paul can work on his USB driver with regard to low power stuff but don't expect anything soon from him on this front. But that example hopefully will get you somewhere in the mean time.

michaelmargolis, 2017-04-15:
Thanks for the prompt reply.  I am using Serial3 to avoid problems with the HID USB driver.  Have you tested using one of the non USB serial ports after wake up?

Michael Margolis

P.S.  I like your library a lot. I will mention it in the next edition of the Arduino Cookbook.

duff2013, 2017-04-16:
Oh I see its the Hardware Serial thats messing up, let me take a look and see whats going on.

duff2013, 2017-04-19:
I think you need ```Serial3.flush();```  before calling ```who = Snooze.hibernate( config_teensy32 );```. I just tried it with Serial1/2/3 and they all worked fine for mulitple wake events. Though you have to change the wakeup pin for Serial2 since its TX line is pin 10.

michaelmargolis, 2017-04-23:
Adding flush before sleeping fixes the problem so am closing this issue.

closed 2017-04-23 by michaelmargolis