duff2013/Snooze #28: Alarm with hour,minute, second params does not wake

michaelmargolis, 2017-04-23:
I am able to trigger alarms passing the epoch for the alarm time but not the hours minutes and seconds.
this fragment illustrates the issue:
```
time_t t = now() + 20; // 20 seconds from now
alarm.setAlarm(t); // this works

alarm.setAlarm(hour(t), minute(t), second(t));  //this does not trigger at the expected time 
```
Have I missed something?

here is the full test sketch
```
#include <TimeLib.h>
#include <Snooze.h>

#define debugPort Serial3

SnoozeAlarm  alarm;

SnoozeBlock snoozeConfig(alarm); 

const int alarmTriggered = 35; // friendly name 

int wakeReason;

void setup() {  
  debugPort.begin(57600); //Tx on pin 8
  debugPort.println("debug serial started\n");
 
  setSyncProvider(getTeensy3Time);
  time_t t = now();
  debugPort.print("time now is "); digitalClockDisplay(t);
  t = t+20;  // next alarm in 20 seconds
  debugPort.print("next alarm will be at "); digitalClockDisplay(t);
  //alarm.setAlarm(t);
  alarm.setAlarm(hour(t), minute(t), second(t));
  debugPort.flush();
  wakeReason = Snooze.hibernate(snoozeConfig);  
}

void loop()
{
  if (wakeReason == alarmTriggered) {
    time_t t = getTeensy3Time();
    debugPort.print("triggered by alarm at");  digitalClockDisplay(t);
    t += 20;
    debugPort.print("next alarm will be at "); digitalClockDisplay(t);
    //alarm.setAlarm(t);
    alarm.setAlarm(hour(t), minute(t), second(t));
    debugPort.flush();
    wakeReason = Snooze.hibernate(snoozeConfig);// wakeReason is module that woke processor
  }
}

void digitalClockDisplay(time_t t)
{  
  char timeStr[12]; 
  sprintf(timeStr, "%d:%02d.%02d", hour(t), minute(t), second(t)); 
  debugPort.println(timeStr);
}

time_t getTeensy3Time()
{
  return Teensy3Clock.get(); 
}
```

MarcianoPreciado, 2017-04-25:
What do you  mean by the "expected time"? Are you anticipating that it will wake when the time is hh:mm:ss during the day, or hh:mm:ss after entering sleep?

duff2013, 2017-04-25:
Absolute time is not supported yet, the function ```alarm.setAlarm(hours, minutes, seconds);``` will fire when (x hours + x minutes + x seconds) have elapsed since the current rtc time when going to sleep.

michaelmargolis, 2017-04-28:
duff, are you saying that setAlarm is functionally the same as the timer methods except it is triggered by the RTC instead of the LPT? I would expect an alarm function driven from a real time clock to work on clock time (like an alarm clock, i.e. to trigger at a given time of day).  I was hoping to suggest the snooze library for people wanting  low power alarm clock functionality but it sounds like setAlarm is not the way to achieve that.

duff2013, 2017-04-28:
You can also just set a variable like ```alarm.setAlarm("time_t value");```.  These two functions work just like an alarm clock in that you set the alarm to some time in the future and it wakes the Teensy.

I think the time_t variable overloaded function ```alarm.setAlarm("time_t value");``` is what you're looking for. If you look at the "TimeTeensy3.ino" sketch you will see that library uses the same way to set the RTC for the correct time "time_t var" so thats a standard way to use time setting.

michaelmargolis, 2017-04-28:
> Absolute time is not supported yet, the function alarm.setAlarm(hours, minutes, seconds); will fire when (x hours + x minutes + x seconds) have elapsed since the current rtc time when going to sleep. 

In my testing, I found that the alarm did appear to  trigger on absolute time, For example calling alarm.setAlarm(Teensy3Clock.get() +60));  triggers an alarm in 60 seconds.

alarm.setAlarm(60);  did not trigger an alarm (the absolute time was on minute past midnight Jan 1 1970);

Is that what you would expect?

duff2013, 2017-04-29:
yes, that is how its programmed:
```
alarm.setAlarm(Teensy3Clock.get() +60));
```
so I see your confusion becasue the other alarm function takes care of ```now()```. This was recently added so no documenting has been done.

michaelmargolis, 2017-04-29:
Thanks for clarifying that.

The method names are what confused me. I guess you can address this with documentation, although if 
it’s not too late because these features have not yet been documented, you may want to consider using naming something like the following.    The last method is not implemented in the current library but would be a useful addition for applications that need to wake up at least once every day.

```
Instead of: setAlarm( time_t alarmTime)

// Set the Period for the RTC to wake the processor
// the interrupt occurs after the given number of seconds has elapsed
void SnoozeAlarm::setRtcTimer( time_t period);

instead of: setAlarm(  uint8_t hours, uint8_t minutes, uint8_t seconds );
// similar to setRtcTimer, period is set using values for hours, min, secs
void SnoozeAlarm::setRtcTimer(  uint8_t hours, uint8_t minutes, uint8_t seconds )

//sets the alarm for the given future absolute time 
void SnoozeAlarm::setAlarm(  time_t alarmTime);

// sets the alarm for the next occurrence of the given hour, minute and second
void SnoozeAlarm::setAlarm(  uint8_t hours, uint8_t minutes, uint8_t seconds )
{
  time_t  alarm = hours * 3600 + minutes * 60 + seconds;
  time_t t = now();
  if (now() < previousMidnight(t) + alarm) {
    // the next alarm is later today
    setAlarm(previousMidnight(t) + alarm);
  }
  else {
    // alarm is for tomorrow
    setAlarm(nextMidnight(t) + alarm);
  }
}
```

duff2013, 2017-05-02:
Let me think about this, I'm looking at the TimeAlarm library as reference now, but if you want to do a pull request please do so, might move things along faster with me:)

michaelmargolis, 2017-05-02:
The key question regards method naming and my proposed  change of functionality for setAlarm(h,m,s).  Are you ok moving the existing functionality for this into  a new method named something like setRtcTimer

It will break sketches using the current implementation of setAlarm(h,m,s)

duff2013, 2017-05-02:
So I guess any RTC function that the user does not have to include the "now()" could be consider a RTC Timer and any function that the user has to provide "now() + some time in the future" could be considered an RTC Alarm.  I have wanted to expand the RTC stuff for a while but havent thought about for some time now. Let me think about this naming change for bit and I'll get back to you soon.

michaelmargolis, 2017-05-02:
The terminology used in the timeAlarms library is that timers are duration based and alarms are triggered at a given time of day. That is the terminology I have used in the method names proposed in my earlier post. In that context, now() plus some period of time is a timer function because the method argument is a duration, not time of day.

duff2013, 2017-05-03:
Yes, that is what I'm looking for, I have been busy these last few weeks so my responses might be sporadic at times. I would like to add the same kind of functionality as timeAlarms library.

duff2013, 2017-07-06:
[michaelmargolis](https://github.com/michaelmargolis)  If you are still following this issue thread I'm finally going to start working on this RTC stuff:)

MarcianoPreciado, 2017-07-06:
I wrote the overloaded setAlarm(time_t) version to eliminate the timer vs alarm confusion, but it seems that it only made it temporarily worse :S. I'm glad to see that it will be wrapped up now. I'd be happy to contribute to the library further, but am rather busy at the moment.

MarcianoPreciado, 2017-07-06:
@michaelmargolis 
If it isn't too late, here is what the implementation looks like in Snooze/utility/SnoozeAlarm.cpp.
```

/*******************************************************************************
 *  <#Description#> Sets an alarm for a specific amount of time once the driver
 *                  is enabled. Is dependent of when the driver is enabled.
 *
 *  @param hours   <#hours description#>
 *  @param minutes <#minutes description#>
 *  @param seconds <#seconds description#>
 *******************************************************************************/
void SnoozeAlarm::setAlarm(  uint8_t hours, uint8_t minutes, uint8_t seconds ) {
    isUsed = true;
    alarm = hours*3600 + minutes*60 + seconds;
    timer_ = true;
}

/*******************************************************************************
 *  <#Description#> Override of original setAlarm function. Works more like a
 *                  standard alarm. Is independent of when the driver is enabled.
 *                  (So long as it is enabled before the alarmTime)
 *
 *  @param alarmTime <#time_t number of specific date & time to set alarm#>
 *******************************************************************************/
void SnoozeAlarm::setAlarm( time_t alarmTime){
    isUsed = true;
    alarm = alarmTime;
    timer_ = false;
}
```

MarcianoPreciado, 2017-07-06:
... and here is what the implementation at the end of SnoozeAlarm::enableDriver() looks like.
```
   ...
    IER = RTC_IER;
    if(timer_) // If setting timer style alarm
        RTC_TAR = rtc_get( ) + ( alarm - 1 );
    else        // else, setting true alarm
        RTC_TAR = alarm - 1;
    RTC_IER = RTC_IER_TAIE_MASK;
}
```

duff2013, 2017-07-06:
Thanks @MarcianoPreciado I'll post what I'm thinking in a bit and we can take it from there.

duff2013, 2017-07-06:
So I'm thinking that I will keep it simple and use two fuctions:
```
    void setRtcTimer( uint8_t hours, uint8_t minutes, uint8_t seconds );
    void setAlarm( time_t alarmTime );
```

Basically just name changes, I'm thinking about adding a third function which will take absolute date&time parameters but still need to figure out how to best do that.

closed 2018-09-04 by duff2013