duff2013/Snooze #30: Snooze Audio Example

becauseinterwebs, 2017-07-08:
Is there an example for using the audio adaptor?  Also, it says that it can wake up the adaptor without waking up the Teensy...does that mean the audio adaptor would still be able to detect input (from the mic or line-in jack, for example) and then wake the Teensy?  Not sure I understand how that works ;)  Thank you!

duff2013, 2017-07-10:
I'll take a look at this tomorrow.

duff2013, 2017-07-11:
The Audio Adapter currently doesn't have a way to wake the Teensy, I'll take a look to see if there is some way. Also I don't know how to put the Audio Adapter into low power mode? Do you know how?

duff2013, 2017-08-19:
The [PRJC Audio Adapter (SGTL5000)](https://www.pjrc.com/store/teensy3_audio.html) most likely cannot wake the teensy. All the signals in deepSleep or hibernate are in a static state so effectively the Audio Adapter is at a stand still.

closed 2020-02-05 by duff2013