duff2013/Snooze #32: Hibernate not able to wakeup

tranphankhanhuyen, 2017-07-25:
Hi, 

I want to implement **Hibernate** but it's not able to wake the system up with a specified button (pin 33 in my sample). My Teensy version is 3.5 . Below is my sample code:

```
// Sample code: Hibernate after 5 seconds
#include <Snooze.h>

SnoozeDigital digital;
SnoozeCompare compare;
SnoozeTimer timer;
SnoozeAlarm alarm;

#if defined(__MK66FX1M0__)
SnoozeBlock config_teensy36(digital, alarm, compare);
#elif defined(__MK64FX512__)
SnoozeBlock config_teensy35(digital, timer, compare);
#elif defined(__MK20DX256__)
SnoozeBlock config_teensy32(digital, timer, compare);
#elif defined(__MK20DX128__)
SnoozeBlock config_teensy30(digital, timer, compare);
#elif defined(__MKL26Z64__)
SnoozeBlock config_teensyLC(digital, timer, lc5vBuffer);
#endif

int sleepTiming = millis();
void setup() {
  Serial.begin(9600);
  pinMode(33, INPUT_PULLUP);
  digital.pinMode(33, INPUT_PULLUP, RISING);
}

void loop() {
  if (millis() - sleepTiming > 5000) {
    Serial.println("Test Hibernate");
    delay(100);
    #if defined(__MK66FX1M0__)
        Snooze.hibernate( config_teensy36 );// return module that woke processor
    #elif defined(__MK64FX512__)
        Snooze.hibernate( config_teensy35 );// return module that woke processor
    #elif defined(__MK20DX256__)
        Snooze.hibernate( config_teensy32 );// return module that woke processor
    #elif defined(__MK20DX128__)
        Snooze.hibernate( config_teensy30 );// return module that woke processor
    #elif defined(__MKL26Z64__)
        Snooze.hibernate( config_teensyLC );// return module that woke processor
    #endif

    setSyncProvider(getTeensy3Time);
    sleepTiming = millis();
  }
  delay(100);
}
```

duff2013, 2017-07-25:
Use the USB Serial Hibernate example if you want to print usb messages. When you put the Teensy into hibernate the USB is disabled and takes time to come back online. Also try putting a debug led blink after wakeing so you know it woke up.

tranphankhanhuyen, 2017-08-01:
Thanks for your response. In my case, it's not only with USB Serial. I have tried with led blink or other output but still not able to wake it up when pressing the button.

closed 2018-11-21 by duff2013

Llamero, 2020-01-21:
I am having a similar issue on a Teensy 3.6 where pins 26, 30, and 33 are unable to wake the device.  I confirmed that these pins were functional by testing if they can detect that the pin was pulled low when not in hibernate.  Here is the test code:

`#include <Snooze.h>
// Load drivers
int interrupt_pins[] = {2,4,6,7,9,10,11,16,21,22,26,30,33};
SnoozeDigital digital;
SnoozeUSBSerial   usb;
Snoozelc5vBuffer  lc5vBuffer;
SnoozeBlock config_teensy36(usb, digital);

void setup() {
  Serial.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
  
  //Confirm that pins 26, 30, and 33 are functional - use pin 2 as a control
  pinMode(2, INPUT_PULLUP);
  pinMode(26, INPUT_PULLUP);
  pinMode(30, INPUT_PULLUP);
  pinMode(33, INPUT_PULLUP);

  //Set LED to flash as output
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWriteFast(LED_BUILTIN, HIGH);

  //Wake USB
  wakeUsb();

  //Configure interrupt pins
  for(int a=0; a<sizeof(interrupt_pins) / sizeof(interrupt_pins[a]); a++){
    digital.pinMode(interrupt_pins[a], INPUT_PULLUP, RISING);//pin, mode, type
    Serial.print("Loading pin ");
    Serial.println(interrupt_pins[a]);
    delay(100);
    digitalWriteFast(LED_BUILTIN, LOW);
    delay(100);
    digitalWriteFast(LED_BUILTIN, HIGH);
  }
}

void loop() {
    int who;
    who = Snooze.hibernate( config_teensy36 );// return module that woke processor
    digitalWriteFast(LED_BUILTIN, LOW);
    wakeUsb();
    while(!digitalRead(2) || !digitalRead(26) || !digitalRead(30) || !digitalRead(33)); //Confirm that pins work by holding them low
    Serial.println(who);
    digitalWriteFast(LED_BUILTIN, HIGH);
    delay(500);
}

void wakeUsb(){
    elapsedMillis time = 0;
    while (!Serial && time < 1000) {
        Serial.write(0x00);// print out a bunch of NULLS to serial monitor
        digitalWriteFast(LED_BUILTIN, HIGH);
        delay(30);
        digitalWriteFast(LED_BUILTIN, LOW);
        delay(30);
    }
    // normal delay for Arduino Serial Monitor
    delay(200);
}
`