duff2013/Snooze #33: Not able to wake up with multiple assigned buttons

tranphankhanhuyen, 2017-07-25:
Hi,

For my sample code below, I assign 2 buttons (33 & 35) to wake the system up when it goes to **sleep**. Still, it freezes every time I press either one of them after the system goes to sleep. It only works when I assign 1 button (33, for example) instead of 2. Below is my sample code:

(My Teensy version is 3.5 and I'm using Snooze 6.2.8)

```
#include <Snooze.h>

SnoozeDigital digital;
SnoozeCompare compare;
SnoozeTimer timer;
SnoozeAlarm alarm;

#if defined(__MK66FX1M0__)
SnoozeBlock config_teensy36(digital, alarm, compare);
#elif defined(__MK64FX512__)
SnoozeBlock config_teensy35(digital, timer, compare);
#elif defined(__MK20DX256__)
SnoozeBlock config_teensy32(digital, timer, compare);
#elif defined(__MK20DX128__)
SnoozeBlock config_teensy30(digital, timer, compare);
#elif defined(__MKL26Z64__)
SnoozeBlock config_teensyLC(digital, timer, lc5vBuffer);
#endif

int sleepTiming = millis();
void setup() {
  Serial.begin(9600);
  pinMode(33, INPUT_PULLUP);
  pinMode(35, INPUT_PULLUP);

  digital.pinMode(33, INPUT_PULLUP, RISING);
  digital.pinMode(35, INPUT_PULLUP, RISING);
}

void loop() {
  if (millis() - sleepTiming > 3000) {
    Serial.println("Test Sleep");
    delay(100);
    #if defined(__MK66FX1M0__)
        Snooze.sleep( config_teensy36 );// return module that woke processor
    #elif defined(__MK64FX512__)
        Snooze.sleep( config_teensy35 );// return module that woke processor
    #elif defined(__MK20DX256__)
        Snooze.sleep( config_teensy32 );// return module that woke processor
    #elif defined(__MK20DX128__)
        Snooze.sleep( config_teensy30 );// return module that woke processor
    #elif defined(__MKL26Z64__)
        Snooze.sleep( config_teensyLC );// return module that woke processor
    #endif

    sleepTiming = millis();
  }

  delay(100);
}
```

Thanks,
Uyen

duff2013, 2017-07-25:
This is because pin 35 is not a wakeup pin.

duff2013, 2017-07-26:
Sorry I was wrong 35 is wakeup source for both sleep, deepSleep and hibernate. I'll take a look at your sketch.

tranphankhanhuyen, 2017-08-01:
Thanks for your response. Then what pin would be the wakeup source? Do I have to reassign my buttons to the correct pins that the Snooze library explicitly recognizes?

duff2013, 2017-08-01:
> Thanks for your response. Then what pin would be the wakeup source? Do I have to reassign my buttons to the correct pins that the Snooze library explicitly recognizes?
Yes, if you want your buttons to wakeup the Teensy from either "deepSleep" or "hibernate" you have to use certian pins that are listed in the examples. "sleep" can use any pin but has much higher power usage.

Snooze will configure the pins right before going into any low power mode and reconfigure to whatever they where before going to sleep when waking. In your example above pin 33 and 35 will be your wakeup pins, you can just use a wire connected to ground to trigger them. I haven't tried those pins yet, been busy with making a driver for the Audio and SPI but I wil soon and see if there is problem.

closed 2018-11-21 by duff2013