duff2013/Snooze #35: deepSleep prevents loading new code?

Erik-k, 2017-08-01:
I'm dealing with an issue where it looks like my Teensy 3.2 is in deepSleep mode (which I put it in) but that turns off the bootloader, and now I can't get any new code loaded on there. I posted the issue on the official pjrc forum:
[https://forum.pjrc.com/threads/45544-Teensy-3-2-no-longer-able-to-upload-code-via-Arduino-IDE?p=149695#post149695](forum post)

Is this a known issue where someone can "lock" themselves out of a Teensy 3.2 by putting it in deepSleep, which turns off the bootloader, which disables loading new code?

duff2013, 2017-08-02:
no i should not, can you:
1. open arduino compile the blink example.
2. unplug the usb cable from computer.
3. hold down the program button on the teensy and plug the usb cable back in.
4. release the program button, the teensy loader should upload the blink example.

Erik-k, 2017-08-02:
I tried that but it doesn't work, the Teensy doesn't even show up on a COM port anymore.

duff2013, 2017-08-02:
Are you sure your cable is good, have you tried it with other Teensies? Did you have anything connected to the Teensy i.e. sensors, motors, etc.. I have about thrity Teensies and this library has been around for 3 years now and never had sleeping a Teensy result in a bricked Teensy.

Erik-k, 2017-08-04:
I took the teensy and cable and used it with a new computer that had brand-new installs of Arduino and the Teensy Loader, and it worked. So it must be something I borked on my other computer. :-( ?

closed 2017-08-04 by Erik-k