duff2013/Snooze #36: Changing interrupt pins?

AndrewCarterUK, 2017-08-08:
Hello,

Is it possible to enable and disable interrupts from different pins between sleeps?

I've tried using multiple digital blocks but I believe the use of static scope has prevented this from working.

Many Thanks,

Andrew

duff2013, 2017-08-09:
Yes, you can have more than one SnoozeBlocks that connects different Digital Drivers which can have different pin wakeups. Can you post a simple example of what you are trying to doing?

AndrewCarterUK, 2017-08-09:
```cpp
#include <Arduino.h>
#include <Snooze.h>

SnoozeAlarm alarm;
SnoozeDigital digital_button_and_papir;
SnoozeDigital digital_button_only;

SnoozeBlock config_full(alarm, digital_button_and_papir);
SnoozeBlock config_no_papir(alarm, digital_button_only);

void sleep_setup(void)
{
  pinMode(HAL_PIN_FUNCTION_BUTTON, INPUT);
  digital_button_and_papir.pinMode(HAL_PIN_FUNCTION_BUTTON, INPUT, RISING);
  digital_button_only.pinMode(HAL_PIN_FUNCTION_BUTTON, INPUT, RISING);

  pinMode(HAL_PIN_PAPIR_INTERRUPT, INPUT_PULLDOWN);
  digital_button_and_papir.pinMode(HAL_PIN_PAPIR_INTERRUPT, INPUT, RISING);
}

void sleep_full(uint32_t seconds)
{
  alarm.setAlarm(seconds);
  Snooze.hibernate(config_full);
}

void sleep_no_papir(uint32_t seconds)
{
  alarm.setAlarm(seconds);
  Snooze.hibernate(config_no_papir);
}

void setup(void)
{
  sleep_setup();

  // sleep_full(1);

  while (1) {
    sleep_no_papir(60);
  }
}
```

I'm trying to create two different sleep functions with the follow functionality:

- `sleep_full(s)` - Sleep until `s` seconds have passed, the PAPIR interrupts or the button is pressed
- `sleep_no_papir(s)` - Sleep until `s` seconds have passed or the button is pressed

The code above works, and the PAPIR will not wake the processor in the sleep loop. However, if I call the `sleep_full(1)` function before hand - the PAPIR will continue to wake the processor in the `sleep_no_papir()` function calls.

This behaviour only happens with `Snooze.hibernate()`, if I change to using `Snooze.sleep()` it behaves as expected.

Thanks!

AndrewCarterUK, 2017-08-10:
Just a note to say that if you think you know what's causing this and can point me in the right direction I'd be happy to submit a pull request or make the library modification myself.

I'm not really sure where the best place to look is.

duff2013, 2017-08-11:
The source code is in the folder '[utility/SnoozeDigital.cpp](https://github.com/duff2013/Snooze/blob/master/utility/SnoozeDigital.cpp)' for pin wakeups. The way these 'drivers' work is they inherit from the SnoozeBlock virtual functions '[enableDriver](https://github.com/duff2013/Snooze/blob/master/SnoozeBlock.h#L357)', '[disableDriver](https://github.com/duff2013/Snooze/blob/master/SnoozeBlock.h#L368)' and '[clearIsrFlags](https://github.com/duff2013/Snooze/blob/master/SnoozeBlock.h#L377)' which get called before going to sleep (enableDriver), after waking from sleep (disableDriver) and if the module needs to clear interrupt flags (clearIsrFlags). These drivers virtual override functions get called by '[sleep](https://github.com/duff2013/Snooze/blob/master/Snooze.cpp#L70)', '[deepSleep](https://github.com/duff2013/Snooze/blob/master/Snooze.cpp#L94)' and '[hibernate](https://github.com/duff2013/Snooze/blob/master/Snooze.cpp#L136)'. Look at the example SkeletonDriverExample.ino for how to make a driver. 

With that said if there is a bug its probably in the disableDriver function that would handle clearing all pin wakeups and reseting the pins back to what they were before sleeping. Hopefully that makes sense. I'll take a look too.

AndrewCarterUK, 2017-08-11:
Could it be [this line in SnoozeDigital](https://github.com/duff2013/Snooze/blob/master/utility/SnoozeDigital.cpp#L294)?

It looks like with SnoozeDigital some of the clean up is done in the ISR and not done for deepSleep/hibernate.

What's the reason for not doing it in these modes?

duff2013, 2017-08-11:
For deepSleep and hibernate the ISR function just disables the pin interrupt as early as possible. 

I did some tests on your example which I had to make some assumptions because not all the info was provided. It worked great for me so maybe its just how your are setting up stuff. Here is what I did.
1. The pins defined are 2, 4, 6
2. Changed INPUT to INPUT_PULLUP
3. Added a blink function to signal when it wakes up.
4. Had to add the loop function to get it compile

The example you gave has no way to tell you it woke up unless your monitoring the current consumption. Another is declaring the wakeup pin ```INPUT, RISING``` but don't you want ```INPUT_PULLDOWN``` and then the signal will pull that pin up hence the RISING part? If its not pulled down won't the pins just float, I guess that depends what you have the wakeup pins connected to? One thing I did test and I'm pretty sure that it works correctly is having mulitple SnoozeBlocks. With my changes I un-commented ```sleep_full(10)``` and it worked as expected. Meaning either pin would wake ```sleep_full(10)``` and only one pin would wake ```sleep_no_papir(60);```

AndrewCarterUK, 2017-08-14:
Were you using hibernate in your tests? All our tests work as long as we're not using the hibernate mode.

We're also monitoring hibernate via current consumption and also using an FTDI serial line.

AndrewCarterUK, 2017-08-14:
Okay, I've done a recording of this test so that you can see the setup.

Here is the complete source code:

```cpp
#include <Arduino.h>
#include <Snooze.h>

#define PIN_BUTTON_1 21
#define PIN_BUTTON_2 22

#define LED 13

SnoozeDigital button_1_digital;
SnoozeBlock button_1_config(button_1_digital);

SnoozeDigital button_2_digital;
SnoozeBlock button_2_config(button_2_digital);

void blink(void)
{
  digitalWrite(LED, HIGH);
  delay(100);
  digitalWrite(LED, LOW);
}

void setup(void)
{
  pinMode(PIN_BUTTON_1, INPUT_PULLUP);
  button_1_digital.pinMode(PIN_BUTTON_1, INPUT_PULLUP, RISING);

  pinMode(PIN_BUTTON_2, INPUT_PULLUP);
  button_2_digital.pinMode(PIN_BUTTON_2, INPUT_PULLUP, RISING);

  pinMode(LED, OUTPUT);

  delay(1000);
}

void loop(void)
{
  Snooze.hibernate(button_1_config);

  blink();

  Snooze.hibernate(button_2_config);

  blink();
}
```

As you see, the program should go from being woken by one button to being woken by another.

The code works properly if you switch `Snooze.hibernate` to `Snooze.sleep` but otherwise the LED can be made to blink by either button (although only by button 1 on the first round).

Here is a [video of the setup](https://www.youtube.com/watch?v=cmvV53LiyXg)

duff2013, 2017-08-14:
Thanks for posting the complete code and the video, I'll test this out today and get back to you.

duff2013, 2017-08-15:
You found a bug! I have confirmed what you are seeing and I have a fix for it in the upcoming update this week. If you want to fix it now you can add the code below to '[utility/wake.h](https://github.com/duff2013/Snooze/blob/master/utility/wake.h#L326)' before the return statement.

```
        mask->PE1 = 0;
        mask->PE2 = 0;
        mask->PE3 = 0;
        mask->PE4 = 0;
#if defined(HAS_KINETIS_LLWU_32CH)
        mask->PE5 = 0;
        mask->PE6 = 0;
        mask->PE7 = 0;
        mask->PE8 = 0;
#endif
```

AndrewCarterUK, 2017-08-15:
Thanks very much, that works for us!

I'm happy with this being closed - but might be worth keeping it open until the fix is in a release.

I'm also happy to submit it as a pull request if you like - just let me know.

duff2013, 2017-08-16:
I already have the fix in my update so I think we can hold off on the pull but if you find anything else feel free to issue a pull request. I'm hoping to wrap up all the updates by friday.

duff2013, 2017-09-06:
I updated [Snooze,](https://github.com/duff2013/Snooze) give it try and let me know.

closed 2018-11-21 by duff2013