duff2013/Snooze #37: Pin compare on Teensy 3.2 pin 12 not working

Erik-k, 2017-08-18:
Has anyone been able to get a compare.pinMode wakeup to work on a Teensy 3.2 on pin 12? It says that pins 11 and 12 can do it but the exact same code works on pin 11 and doesn't work on pin 12, and I tried this across two Teensy 3.2's, including a brand new one that I just bought only to test this. Still only works on pin 11 which is unfortunately also the SPI MOSI default pin.

duff2013, 2017-08-18:
ok, I'll take a look.

duff2013, 2017-08-19:
So it turns out pin 12 is not wakeup pin for compare driver in deepSleep or hibernate. The good news is the Teensy 3.x can use pins 9 and 4 now with my upcoming update.  The Teensy LC can only use pin 11. I was hoping to push this out tonight but I still have a bunch of things to fix for using Snooze with the Audio library.

duff2013, 2017-09-06:
I update [Snooze](https://github.com/duff2013/Snooze) give it try and let me know if the pin compare works.