duff2013/Snooze #39: Example doesn't compile with Teensy LC

benkrasnow, 2017-09-18:
I downloaded a fresh Arduino 1.8.4, added Teensyduino 1.39, and tried to compile Examples/Snooze/deepsleep/deepsleep_all_wakeups, but had an error.  The problem is localized to use of keywords in SnoozeCompare.cpp that are not available in Teensy LC.

I commented references to IRQ_CMP2 and LLWU_CMP2_MOD, and everything compiles and the timer wakeup works (I'm not using the comparator).

I'm new to github, but would be willing to try fixing this.  I really appreciate your work to produce this library!

Snooze\utility\SnoozeCompare.cpp:195:42: error: 'IRQ_CMP2' was not declared in this scope

     if ( mode == VLPW ) NVIC_ENABLE_IRQ( IRQ_CMP2 );

Snooze\utility\SnoozeCompare.cpp:170:57: error: 'LLWU_CMP2_MOD' was not declared in this scope

         if ( mode >= LLS ) llwu_configure_modules_mask( LLWU_CMP2_MOD );

duff2013, 2017-09-19:
Yep, I didn't check the LC. I'll have update this week sometime.

nobodyguy, 2017-10-23:
How does it work on Teensy LC? Does the used ARM chip have usable RTC?

duff2013, 2017-10-23:
This issue was about the Compare Hardware on the Teensy LC but as far as I know there is no RTC hardware on the LC just the logic so after power is removed it will lose the time.