duff2013/Snooze #40: Wire conflicts

Adrianotiger, 2017-10-04:
I am using a i2c wire device.
With the last version, I am not able to compile it anymore (multiple definitions of Wire).

I saw that SnoozeAudio has this include:
"Wire.h"

If I replace it with "#include <i2c_t3.h>"
it works.

I know, it should remain compatible... But I was not able to compile it with Wire.h.

duff2013, 2017-10-04:
What version of Snooze are you using and can post an example that shows this problem?

Adrianotiger, 2017-10-05:
Sure! :D
`
#include <TimeLib.h>
#include <Snooze.h>
#include <i2c_t3.h>

SnoozeTimer timer;
SnoozeBlock config(timer);

void setup() {
  Wire.begin(I2C_SLAVE, 0x66, I2C_PINS_18_19, I2C_PULLUP_EXT, 100L * 1000L);
  timer.setTimer(1000);
}

void loop() {
  Snooze.hibernate( config, LLS );
}
`

I used the version of yesterday (4 oct 2017), since the one from Stoffregen doesn't compile if I don't want the USB. 

I don't have a real solution.
A workaround would be to write something like:
`
#ifdef I2C_T3_H
#include <i2c_t3.h>
#else
#include "Wire.h"
#endif
`
But this works only if i2c_t3 was included in the project before snooze.h.

Ah, I am using Teensy 3.2.

duff2013, 2017-10-05:
Ok I see whats going on now, just delete: #include "Wire.h" in SnoozeAudio, it's not needed. This fix will be in the upcoming update too.