duff2013/Snooze #43: Support for SLEEP_MODE_IDLE on Teensy 3

arkhipenko, 2017-12-16:
Hi,

I would like to implement support for low power sleep (similar to SLEEP_MODE_IDLE of the Arduino) on Teensy to make it part of the TaskScheduler library
(https://github.com/arkhipenko/TaskScheduler)
The Teensy "avr/sleep.h" does not seem to have have any code actually, and "avr/power.h" is simply empty. 
I have looked at the Snooze library and tried to use the Timer initiated power sleep, but it seems to be working differently then Arduino.

TaskScheduler uses SLEEP_MODE_IDLE to place Arduino into low power mode for approximately 1 ms. It is awaken back by the timer interrupt responsible for updating millis() counter. This way I am achieving 1ms scheduling resolution. Arduino does not power down any peripherals while in SLEEP_MODE_IDLE, so USB and other communications continue to run normal.

Teensy seems to be behaving differently. 
```
class definition:
#ifdef CORE_TEENSY
	SnoozeTimer *timer;
	SnoozeBlock *config;
#endif //CORE_TEENSY

constructor:
#ifdef CORE_TEENSY
	timer = new SnoozeTimer;
	config = new SnoozeBlock(*timer);
	timer->setTimer(1);
#endif //CORE_TEENSY

actual sleep:
#ifdef CORE_TEENSY
	Snooze.sleep(*config);
#endif //CORE_TEENSY
```
causes my Tennsy 3.5 periodically connect and disconnect, so I can't even monitor what is going on with the device. 

Is there a sleep function that is close to AVR's SLEEP_MODE_IDLE mode?

Your help is greatly appreciated. 
Thanks.
Anatoli

arkhipenko, 2017-12-16:
I created a "teensy_idle" branch for your reference of my changes if you have a minute:
This sums up the TEENSY related updates pretty well:
https://github.com/arkhipenko/TaskScheduler/compare/teensy_idle?diff=split&name=teensy_idle

duff2013, 2017-12-16:
As I said in the Teensy forum that just calling wifi will only pause the processor until interrupt happens, if you want it to pause for set amount of time you have to disable the systick and usb interrupts at a minimum and use a timer to wake in the amount of time you want to sleep.

arkhipenko, 2017-12-16:
Thank you for your help!

closed 2017-12-16 by arkhipenko