duff2013/Snooze #46: Set Timer w/ Microseconds?

ploderup, 2018-02-20:
Considering the Teensy's clock has sub-microsecond length periods, it'd be nice if the timer could be set for shorter than a millisecond. Is there any simple work-around for this?

duff2013, 2018-02-21:
It depends on what sleep mode you are using, 'DeepSleep' and 'Hibernate' would need an external pin trigger but 'Sleep' could use the Teenyduino's 'Interval Timer' which is not implemented through Snooze.

ploderup, 2018-02-21:
Okay, I'll look into using the interval timer. Any idea how snoozing
affects writes to an SD card over SPI? I saw some comments regardings
problems with Serial.println(). Any fixes you're aware of for that (aside
from waiting)?

Best,
ploderup

duff2013, 2018-02-22:
SD should work much quicker to work after waking from sleep mode than Serial. Nothing special should have to be done to make it work.

ploderup, 2018-02-22:
> Actually Interval Timer won't work now that I think about it because
> during 'Sleep' the bus clock is at 2MHz so the Interval Timers won't fire
> at the time you set. I would look into getting external timer that has a
> pin toggle you can connect to a Teensy pin.

Is there a way to sleep at full clock speed? If not, I might just see if I
can't drop power consumption by dropping the clock frequency a little.

duff2013, 2018-02-22:
There is a function in Snooze Called 'Idle' that puts the cpu into wait mode until any interrupt happens which is usually would be systick interrupt every millisecond . You can use it for spin loops where you are waiting for something. It will lower the Teensy power usage and not side effect because the cpu and bus are not reduced.

closed 2020-02-05 by duff2013