duff2013/Snooze #48: Teensy LC: Pin wakeups on pins 2, 7, 9 not working

MatthewStephenson, 2018-02-24:
The README states that for Teensy LC, Digital pins: 2,6,7,9,10,11,16,21,22 should work as wakeup sources. However, having tested them, I found that pins 2, 7, and 9 don't wake the CPU either from deepSleep or hibernate.

duff2013, 2018-12-30:
@jacky4566 you can always give a pull request to fix things too. Thats how open source works.

jacky4566, 2019-01-01:
Sorry @duff2013 didnt meant to sound like a jerk. 
I am just confuse where the original number set came from...

Working on a fix now.

duff2013, 2019-01-02:
Ok no problem I'm going to look at this over the weekend, I see you did a pull thanks! There are few more places that need to be touched also before merge your pull.

jacky4566, 2019-01-18:
No worries Duff!
So this week I tested did some science and test the pins. Here is the results

PIN | LLWUP | 48 QFN Pin | Teensy LC pin | Works?
-- | -- | -- | -- |
PTD0 |   | 41 | 2| N
PTD4 | P14 | 45 | 6 | Y
PTD2 |   | 43 | 7 | N
PTC3 | P7 | 36 | 9 | Y
PTC4 | P8 | 37 | 10 | Y
PTC6 | P10 | 39 | 11 | Y
PTC5 | P9 | 38 | 13 | Y**
PTB0 | P5 | 27 | 16 | Y
PTD6 | P15 | 47 | 21 | Y
PTC1 | P6 | 34 | 22 | Y

** (Should put in a note about using pullups on the Built-in LED)

Test Code:
```
#include <Snooze.h>

SnoozeDigital digital;
SnoozeBlock config_teensyLC(digital);

int testPin = 2;

void setup() {
  int who;
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  digital.pinMode(testPin, INPUT_PULLUP, RISING);//pin, mode, type
  who = Snooze.deepSleep( config_teensyLC );
  digitalWrite(LED_BUILTIN, HIGH);
  while (!Serial);
  elapsedMillis time = 0;
  while (!Serial && time < 1000) {
    Serial.write(0x00);// print out a bunch of NULLS to serial monitor
    delay(30);
  }
  Serial.printf("Pin: %i | Who: %i", testPin, who);
}
```

duff2013, 2020-02-05:
Fixed

closed 2020-02-05 by duff2013