duff2013/Snooze #49: Current with snooze and TSI

Alan-NTHU, 2018-06-30:
On T3.6  (not sure, but probably same to other board)
When using touch pin to wake, it tooks a significant 60mA more currency at hibernate/sleep ,
which is, not good for sleeping project at all. It seems the TSI is continuously back to back scanning in the setting?

Is it possible to set the LPT counter overflow much lesser so it saves power?

I currently use snooze with timer and wake up each 25ms for a touchRead to detect a touch and (break;)
and this case current is much lower(<1mA) at my meters

duff2013, 2018-06-30:
Hi, I'll check this out but from what I remember the T3.6 Hibernate current using TSI was in line with other wakeup methods. Also the T3.6 TSI uses the LPT (Low Power Timer) so you can't use LPT as wakeup source along with the TSI. 

You can change the scan time a couple of ways but the easiest is in SnoozeTouch.cpp in function enableDriver find at the bottom of the function "LPTMR0_CMR=1" and change that number.

duff2013, 2018-06-30:
Yes any external hardware will add current, can you try just bare T3.6 and touch and see what current you measure. That way we can be one the same page.

Alan-NTHU, 2018-06-30:
I just did another test with your latest library and example Hibernate_all_wakeups and bare T3.6 
with block line changed to : SnoozeBlock config_teensy36(touch);
the hibernate with touch draws near 95uA and ~200uA with SD, which is much smaller than my 0.8mA
strange, seemed to work well, possibly my device with PCB shield draws current with unknown reason , maybe it's caused by large surface designed to use for touch? which a naked board does not have. I will test that later. But touch sense are potentially use with large pads( are they?)

I did the same test couple of days ago, as I remember snoozing with touch,
and plugging in and off the built-in SD to see currency difference.
And remember the results were near 60-70mA.SD existence draws 1~2x mA depends on state.
but it was with the previous version without Sdcard block yet, could already be fixed?:\

duff2013, 2018-07-01:
95uA sounds more like it with a bare Teensy and 200uA with the SD Card inserted. I haven't explored if the touch pad size effects the current but try different sizes and let me know. Any signal lines from your shield to the Teensy can cause current draw if the pin is not configured correctly when going to any of the sleep modes. Also try to incorporate a Mosfet power switch to any hardware so you can kill the power if feasible in your project though any signal lines like SPI or I2C still have to be dealt with.

Alan-NTHU, 2018-07-01:
Yes, the other hardware is controlled by N-channel mos on my board, the build-in SD is however excluded, thanks to your SdCard block.
and signal lines are turn to high/low to make sure no voltage different and minimun current leak.
planning to add resister and logic IC to further prevent leak on signal lines.

duff2013, 2018-07-03:
I'm going to close this unless you are still having problems with the Touch sleep wake?

Alan-NTHU, 2018-07-03:
thanks for your replies :) 
if I have further discovery about touch&snooze I'll let you know 👍

closed 2018-11-21 by duff2013