duff2013/Snooze #50: Snooze prevents loading new code?

fahdovski003, 2018-07-31:
When using the Snooze Library, I have the following error when trying to upload new code :

```
Arduino: 1.8.5 (Linux), TD: 1.42, Board: "Teensy LC, Serial, 48 MHz, Smallest Code with LTO, US English"

Sketch uses 13612 bytes (21%) of program storage space. Maximum is 63488 bytes.
Global variables use 3404 bytes (41%) of dynamic memory, leaving 4788 bytes for local variables. Maximum is 8192 bytes.
No Teensy boards were found on any USB ports of your computer.
Please press the PROGRAM MODE BUTTON on your Teensy to upload your sketch.
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
It seems that the port disappears because of the hibernation :
`/home/exia003/Downloads/arduino-1.8.5/hardware/teensy/../tools/teensy_post_compile -file=XbeeNode.pde -path=/tmp/arduino_build_52483 -tools=/home/exia003/Downloads/arduino-1.8.5/hardware/teensy/../tools -board=TEENSYLC -reboot -port=/sys/devices/pci0000:00/0000:00:0c.0/usb1/1-3 -portlabel=(null) -portprotocol=(null) `

if i comment this code ` Snooze.hibernate( config_teensyLC ); //teensy go to hibernate mode for LOOP_DELAY `
then it works again but only if i keep the reset button pressed while trying to upload the sketch several time until it succeeds.

fahdovski003, 2018-07-31:
To be able to upload , you need to :
1- disable the teensy loader automatic mode
2- press the reset button on the teensy board
3- upload the sketch with the arduino ide.

duff2013, 2018-07-31:
Yes this normal behavior. When putting any of the teensy's to sleep, to upload new code you have to push that program button on the teensy when the Arduino IDE is done compiling.

closed 2018-08-01 by fahdovski003