duff2013/Snooze #52: 4.5mA drain in any sleep mode

pldubouilh, 2018-10-10:
Hi, thanks for the great library !

A quick question - I don't seem to be able to go under 4.5mA on my teensy 3.1. Using the (sleep/deep-sleep/hibernate) examples, I'm always in the 4~5 mA range... Any clue where I should investigate ?

I usually use platformio, but I get the exact same results using the recommended arduino+teensyduino way. 

Setup wise, I don't have the crystal soldered (I commented out everything time/rtc-related in the example), and I'm testing the board directly with a hyelec multimeter from the 5v of an USB port on my Vin.

Cheers !

duff2013, 2018-10-10:
Is there anything connected to the Teensy? Also can you explain a little more on how you are measuring the current with that multimeter?

pldubouilh, 2018-10-11:
Nothing is connected, simply using jump wire to connect to the teensy. I also tried feeding 3.3v on the 3.3v input but same result.

My meter setup is fairly simple ; 5v <> meter <> teensy <>  gnd. The meter seems to work alright, it reads properly 500uA when I replace the teensy by a 10kOhm resistance. 

The teensy had a busy-ish past, maybe it was (ab)used a bit too much ? I'm not quite convinced though, any other program works just fine.

pldubouilh, 2018-10-11:
I ordered a new teensy, should arrive by tomorrow, I'll keep you posted if it's working on the new hardware :)

pldubouilh, 2018-10-12:
Alright the teensy 3.2 I received works just fine, no more mA current drain while asleep... Something must be off with the other one.
Thanks, and sorry for the noise :)

closed 2018-10-12 by pldubouilh