duff2013/Snooze #54: Audio library doesn't work after wakeup

yidiwang21, 2018-11-14:
I'm using Teensy 3.6 and the output from the microphone is connected to Pin 16 (adc). I also tried to use SnoozeAudio by declaring usingADC(16) in the setup. After the teensy woke up, the recording queue was still unavailable.

duff2013, 2018-11-14:
Can you post a small program that produces what you are talking about? Also are you using the latest version of Snooze not the one that ships with Teensyduino?

yidiwang21, 2018-11-14:
Yes, I am using the latest version of Snooze. Audio recording works fine when snooze is disabled. 

`SnoozeAlarm alarm;
SnoozeAudio audio;
//SnoozeTimer timer;
SnoozeBlock config_teensy36(alarm, audio);
//SnoozeBlock config_teensy36(timer);

#define SLEEP_MODE_ENABLED
// #define AUDIO_DEBUG
// #define BLE_DEBUG

#define SLEEP_TIME_HR	0
#define SLEEP_TIME_MIN	0
#define SLEEP_TIME_SEC	10	// for testing

#define EX_LED_PIN		31

void ledBlink(int cnt) {
	for (int i = 0; i < cnt; i++) {
		digitalWrite(EX_LED_PIN, HIGH); delay(500);
		digitalWrite(EX_LED_PIN, LOW); delay(500);
    }
}

void setup() {
	pinMode(EX_LED_PIN, OUTPUT);
	Serial.begin(9600);
	BLE._setup();
	_SD._setup();
	AudioRecorder._setup();
	Serial.println("**********************************************");
	ledBlink(2);
	//timer.setTimer(30000);
	alarm.setRtcTimer(SLEEP_TIME_HR, SLEEP_TIME_MIN, SLEEP_TIME_SEC);	// hours, minutes, seconds
	audio.usingADC(16);
}

void loop() {
	// SIM_SCGC6 &= ~SIM_SCGC6_I2S;
  	int who;
  	who = Snooze.deepSleep(config_teensy36);  // get into sleep for 10 min until wakeup
  	// SIM_SCGC6 |= SIM_SCGC6_I2S;

	if (who == 35) {
	  	Serial.println("**********************************************");
    	    delay(1000);
		
		ledBlink(1);
		AudioRecorder.audioRecording();
    	delay(1000);
	}
}
`

And here is my recording functions:
`AudioInputAnalog         audio_adc;   // using adc0 of Teensy 3.6
AudioRecordQueue         audio_queue;
AudioConnection          patchCord2(audio_adc, audio_queue);

File audio_rec;

AudioClass::AudioClass() {
    recordingMode = 0;
}

void AudioClass::_setup(void) {
    delay(100);
	Serial.println("# Teensy Audio AudioMemory()");
    AudioMemory(60);
}

void AudioClass::startRecording(void) {
    // audio_queue.update();
    if (SD.exists("save.raw")) {
    Serial.println("# Existing file removed");
        SD.remove("save.raw");
    }
    audio_rec = SD.open("save.raw", FILE_WRITE);
    if(audio_rec) {
        Serial.println("# Start recording");
        Serial.print("# Queue available ?"); Serial.println(audio_queue.available());
        audio_queue.begin();
        recordingMode = 1;
    }
}

void AudioClass::stopRecording(void) {
    Serial.println("# Stop recording");
    Serial.print("File size: "); Serial.print(audio_rec.size()); Serial.println(" bytes");
    audio_queue.end();
    if (recordingMode == 1) {
        while (audio_queue.available() > 0) {
            audio_rec.write((byte*)audio_queue.readBuffer(), 256);
            audio_queue.freeBuffer();
        }
        audio_rec.close();
    }
    // audio_queue.clear();
    recordingMode = 0;
}

void AudioClass::continueRecording(void) {
    if (audio_queue.available() >= 2) {
        byte buffer[512];
        memcpy(buffer, audio_queue.readBuffer(), 256);
        audio_queue.freeBuffer();
        memcpy(buffer + 256, audio_queue.readBuffer(), 256);
        audio_queue.freeBuffer();
        elapsedMicros usec = 0;
        audio_rec.write(buffer, 512);  //256 or 512 (dudes code)
#ifdef AUDIO_DEBUG
        Serial.print("SD write, us=");
        Serial.println(usec);
#endif 
    }
}

void AudioClass::audioRecording(void) {
    startRecording();
    unsigned long time = millis();
    while (millis() - time < REC_TIME) {
        continueRecording();
    }
    stopRecording();
}`

duff2013, 2018-11-16:
A couple of things: 
1. USB and deepSleep or hibernate don't really play nice together. If you want to use the usb there is Snooze Usb driver and example but reconnecting usb after sleep takes a few seconds.
2. I would separate the SD and Audio when testing with sleeping your teensy. Once you get both working you can combine them later.
3. With the Audio library split that into making sure your mic works and then make sure the Queue works.

I can say that I have successfully used all those components you have with my library. I'll try to help but lets take it one issue at a time.

duff2013, 2018-12-06:
Are you still have problems?

closed 2020-02-05 by duff2013