duff2013/Snooze #55: sleep, deepsleep and hibernate modes

MungoBerry, 2018-12-01:
Hi,

What's are the difference between sleep, deep sleep and hibernate modes ?
Then, I cannot put teensy 3.2 in sleep for 5 minutes, what's the maximum sleep period we can have ?

Thank you.

duff2013, 2018-12-01:
sleep, deepSleep and hibernate are different levels of putting the Teensy into low power modes with deepSleep and hibernate being the same mode but hibernate puts the USB regulator in low power mode only able to supply limited current to the 3.3v out on the Teensy.

To sleep for 5 min use the rtc (alarm driver)

LoKKeR96, 2018-12-20:
What kind of sleep mode is used when RTC (alarm driver) is used?

duff2013, 2018-12-21:
RTC works with all of them.

LoKKeR96, 2018-12-21:
> RTC works with all of them.

Could you help me out with how do I set the sleeping mode for RTC? I didn't find it in the homepage of the project.

duff2013, 2018-12-21:
Take a look at this [example](https://github.com/duff2013/Snooze/blob/master/examples/deepsleep/deepSleep_all_wakeups/deepSleep_all_wakeups.ino), it shows all the wake ups including the RTC.

LoKKeR96, 2018-12-21:
> Take a look at this [example](https://github.com/duff2013/Snooze/blob/master/examples/deepsleep/deepSleep_all_wakeups/deepSleep_all_wakeups.ino), it shows all the wake ups including the RTC.

Yeah thanks that's what I am already doing to set up the RTC sleep. I know that. What I am asking is how to change the sleep mode RTC uses. Like say I want hibernation instead of deepSleep when RTC is used to sleep. Is that possible? If not then I'd like to know what sleep mode is used when the RTC routine for sleeping is used. Thanks

duff2013, 2018-12-21:
The RTC works exactly the same in any sleep mode (sleep, deepSleep, hibernate).

closed 2020-02-05 by duff2013