duff2013/Snooze #60: Millis counter not correct with digital and timer

jacky4566, 2019-01-18:
I know its a bit picky but the millis timer doesnt seem to update correctly when woken from another source besides timer.
Looking at the code we are storing a value, sleeping, then adding that value to the systimer?

So no actual counting is happening while sleeping?

If you can make any suggestions to improve I would be happy to code and test it.

duff2013, 2019-01-25:
millis doesn't work because the systick is off during sleep. 

The LPTMR is only 16 bits so it has limited amount space to increment so if you sleep a long time it will roll over and you won't know unless periodically wake to count the roll overs.

jacky4566, 2019-01-25:
If you are using the LP timer then you won't ever be sleeping more than 65536ms anyway. This would never overflow the systick more than once. 

What I would like to do is read LPTMR0_CNR when teensy is woken by another source.

I tried to modify the isr with this function but still not getting a value.

```
void SnoozeTimer::isr( void ) {
    if ( !( SIM_SCGC5 & SIM_SCGC5_LPTIMER ) ) return;
    actualSleepTime = LPTMR0_CNR;
    systick_millis_count += lptmrUpdateSystick;
    LPTMR0_CSR = LPTMR_CSR_TCF;
}
```

This is a very very low priority thing Duff. Just a push in the right direction would be helpful. 
Your work on Snooze is already awesome sir!

duff2013, 2019-01-25:
There is couple of things off the top of my head, is **actualSleepTime** volatile? It has to be if in an interrupt handler. The LPTMR has to enabled to read the register, I assume you are using the Timer driver for that? Also it won't get passed the if statement if not using the Timer driver. If all that is met then you should be able to read that register, I'll take a look soon because now looking at it my code, updating ```systick_millis_count``` could be flawed, should be checking the actual count not what it is configured for.

closed 2020-04-15 by duff2013