duff2013/Snooze #62: usbSerial

BufYank, 2019-03-05:
I am using a device where I have a program that continually plots a graph with frequency and temperature. I want to place the device to sleep to save power and then wake it up. I have used deepSleep and hibernate, which does great for saving power, but what happens is the USB serial communication is interrupted and my program hangs up. Requires a complete restart of the program.

Is there a way to keep the USB communication going?

I see there is a usbDriver (SnoozeUSBSerial usb;), but how do I implement it? Not even sure if this would fix my problem.

duff2013, 2019-03-05:
What is the teensy connect to through usb, computer?

BufYank, 2019-03-06:
Yes. The teensy is connected and powered via USB to a computer. The program/application is on the computer and is meant to communicate with the teensy 3.2.

duff2013, 2019-03-06:
What OS, Windows can have issues with USB Serial. With that being said look at the USBSerial driver and try the example and see if works. If it does then try add that code to your sketch. 

The issue might be that when you go into those sleep modes the USB connection to the computer is dropped. Your program will have to recognize this and look for a reconnected USB after the Teensy wakes up, but that takes seconds once the Teensy wakes up. A lot of computer programs don't expect this drop of the USB and don't handle it so I can't really help there. 

USB and sleeping Teensy has always been and issue. One thing you can always do is use a USB to Serial converter cable and use the the Teensy Serial ports instead. Then these issues are mitigated.

BufYank, 2019-03-06:
It's Windows 10. The more I have been thinking about this, I've been realizing it's on the program side. Like you said, it would have to recognize a disconnect and then pick up where it left off when the connection is re-established. Once the teensy goes to sleep, the program doesn't see the COM port at all. Which is telling me that Windows has disconnected.

I have tried:
SnoozeUSBSerial usb;
SnoozeBlock config1(usb, timer);
Snooze.deepSleep (config1);

I haven't tried:
`elapsedMillis time = 0;
    while (!Serial && time < 1000) {
        Serial.write(0x00);// print out a bunch of NULLS to serial monitor
        digitalWriteFast(LED_BUILTIN, HIGH);
        delay(30);
        digitalWriteFast(LED_BUILTIN, LOW);
        delay(30);
    }`

I don't think this will fix the problem though. As when usb connection is lost, it can't be re-established until I close out the program and then open it again. The program will actually see the USB device when the teensy wakes but the program is locked up and doesn't start off where it left.

I haven't been able to try the USB to Serial option, I will have to purchase an adapter.

closed 2019-03-07 by BufYank