duff2013/Snooze #64: Can't understand how to suspend timer interrupts

Sciurusvulgaris, 2019-05-17:
I'm playing a WAV file from beginning to end in my main program loop, then go to sleep.  So at the next timer interrupt, the code should start again at the next line, which is the beginning of the loop.  
However, this doesn't work as I intended, because the timer interrupts seem to be interrupting the following loop:
  playWav1.play(wavFile[4]);
  delay(5);   // Brief delay to read WAV info.
  while (playWav1.isPlaying()) {
    }    // Wait for the file to finish playing.
It seems to me that I need to suspend interupts between waking and the sleep command, but I can't figure out how to do that in this library.  Help would be greatly appreciated.

lawnchairsociety, 2019-05-18:
@Sciurusvulgaris Can you provide the code snippet where you are calling the sleep/deepSleep/hibernate? Also, what does your SnoozeBlock and config look like?

Sciurusvulgaris, 2019-05-18:
Hi, thanks for looking at this.  I've attached the code here.  I've shortened it by removing distractions, but hope I've left in eveything relevant!
[code.txt](https://github.com/duff2013/Snooze/files/3193766/code.txt)

lawnchairsociety, 2019-05-21:
@Sciurusvulgaris  Sorry for the delay in responding. Are you getting a second flash from the LED prior to the sleep starting or is it sleeping in the middle of the while loop?

THanks!

duff2013, 2019-05-21:
I'll check this out, I'm in the middle of adding support for the upcoming Teensy 4.0 which means that I have to decouple all the hard coded T3.x/LC stuff into its own HAL. Its major undertaking that is not going so well currently :( Plus I have a plethora issue like this to look at so bare with me.

Sciurusvulgaris, 2019-05-22:
> @Sciurusvulgaris Sorry for the delay in responding. Are you getting a second flash from the LED prior to the sleep starting or is it sleeping in the middle of the while loop?

I get the second flash before sleeping.  Of course then the board is unrecognised by WIndows 10/ARduino IDE, so recovering is a little problematic, but I've managed it so far by loading the simplest possible sketch while the board is briefly awake.

> I'll check this out, I'm in the middle of adding support for the upcoming Teensy 4.0 which means that I have to decouple all the hard coded T3.x/LC stuff into its own HAL. Its major undertaking that is not going so well currently :( Plus I have a plethora issue like this to look at so bare with me.

I can completely relate to that, sympathies.  Whatever you can do will be welcomed by one and all.  I'll keep struggling, but in my case it's a bit like give a monkey a typewriter.