duff2013/Snooze #66: I found a strange code in SnoozeDigital.cpp

AstroTaka, 2019-07-29:
I am using Teensy 3,2, Arduino IDE 1.8.9 and Teensyduino 1.46 combination.
I checked "deepSleep_all_wakeups" example code.
The example has "`digital.pinMode(21, INPUT_PULLUP, RISING)`".
So it should wakeup by RISING edge of Pin 21.
But my teensy wakeups by FALLING edge of Pin21.
So I checked a code of Snooze library.

I found a strange code in SnoozeDigital::enableDriver at 
 "`llwu_configure_pin_mask( pinNumber, pin_mode );`".

I checked llwu_configure_pin_mask function.
The 2nd argument of the function should set "`RISING`", "`FALLING`" or "`CHANGE`".
But original code set "`INPUT`", "`INPUT_PULLUP`" or "`INPUT_PULLDOWN`".

So I changed the code as the following.

`llwu_configure_pin_mask( pinNumber, pin_type );`

And then, I checked modified code and it works fine at my environment.

And I found a code for Teensy LC.
It it same situaion of 3.2 I think.
Please check it.

I checked the latest Snooze v6.3.4.
The part of Teensy 3.2 and LC has the same situation.
(The llwu_configure_pin_mask function is commented out)
Please check it also...

duff2013, 2019-07-30:
Thanks for reporting I'm going to work on Snooze in the next week or so and I'll update this thread when Iook into this.

AstroTaka, 2019-07-30:
OK, Thanks!

duff2013, 2020-02-05:
Yep your right that was an issue, my latest update doesn't have this fix but the next one soon will.

closed 2020-04-15 by duff2013