duff2013/Snooze #69: Teensy 4.0

mbatesole, 2019-12-04:
It appears Snooze/src/hal/TEENSY_40/ is missing the appropriate header files: 

#include "SnoozeTimer.h"
#include "SnoozeAlarm.h"
#include "SnoozeTouch.h"
#include "SnoozeCompare.h"
#include "SnoozeDigital.h"
#include "SnoozeAudio.h"
#include "SnoozeUSBSerial.h"
#include "Snoozelc5vBuffer.h"
#include "SnoozeSPI.h"

duff2013, 2019-12-06:
Yes Teensy 4.0 is not working, I have hope to work on this soon but other things keep getting in my way.

HanJ67, 2020-03-16:
When can we expect a version for Teensy 4.0 to go to sleep and wakeup via interrupt IO

duff2013, 2020-03-16:
I've been working on it stay tuned I'll have a beta hopefully in a week.

HanJ67, 2020-03-16:
Oh great, thanks a lot, really appreciated..

HanJ67, 2020-04-16:
Hi,
Yesterday I tried the updated lib on a Teensy 4.0 environment and got a compiler error on hal.c. I had not enough time to search for the reason. It should work now, right?
thanks and best regards, han

duff2013, 2020-04-16:
I'm working on it, should have another beta out by the weekend.

HanJ67, 2020-04-28:
Works perfectly.....thanks a lot....

duff2013, 2020-04-28:
Glad to hear going to close this issue then.

closed 2020-04-28 by duff2013