duff2013/Snooze #71: Teensy LC does not wake up from on digital interrupt

darianbjohnson, 2020-01-02:
I am having a hard time getting my Teensy LC to wake up (using pins 21 or 22, pin to ground to wake). Using the sleep, deep sleep, and hibernate examples. 

Maybe I'm missing something here. I load the example. Teensy goes to sleep/deepSleep/Hibernate.

I bring pin 21 (or 22) to ground. 

Nothing.

Am I doing something bone-headed here? Seems like it should work. 

Thanks

duff2013, 2020-01-17:
Not sure can you post code that shows this not working, try to make the code concise as possible.

duff2013, 2020-02-05:
I just updated Snooze only deepSleep and hibernate work so far you can download [here](https://github.com/duff2013/Snooze/releases/tag/v6.2.5), let me know if you have any problems.

darianbjohnson, 2020-02-06:
Thanks; I'll test and report back if there are issues.

closed 2020-02-06 by darianbjohnson