duff2013/Snooze #74: Snooze for Teensy 3.6 is incompatible with Teensyduino 1.50

Llamero, 2020-02-04:
I upgraded to Arduino 1.8.11 and Teensyduino 1.50, and was unable to get Snooze to run, even the example blocks of code.  The Teensy 3.6 would enter the sleep state but never wake to timer, RTC, or digital (did not try the other wake options).  Reverting to Arduino 1.8.10 and Teensyduino 1.48 resolved the issue and all of the examples work.  There is no mention of Snooze in the update logs for 1.49 and 1.50, so not sure where to begin to find the source of this issue.

duff2013, 2020-02-05:
I just updated Snooze only deepSleep and hibernate work so far you can download [here](https://github.com/duff2013/Snooze/releases/tag/v6.2.5), let me know if you have any problems.

duff2013, 2020-04-13:
I'm going to close this issue if there is still a problem than let me know.

closed 2020-04-13 by duff2013