duff2013/Snooze #78: Support for mk20dx128 (Teensy 3.0)?

florind, 2020-03-23:
Hi, this is perhaps an odd request but I would like to know how difficult would be to adapt Snooze for the mk20dx128 (older Teensy 3.0)? 
I tried to add a TEENSY_30 src with the changed ```#if defined MK20DX128 ``` and deleted the IRQ_CMP2 references but my Teensy 3.0 doesn't seem to wake up after going to sleep. 

Thank you!
florin

duff2013, 2020-03-23:
What are trying to use to wakeup, compare? I'm not sure if the Teensy 3.0 has that though I haven't looked at Teensy 3.0 in 7 years. I can say I'm not going to support Teensy 3.0 since Paul dropped support also. Why not just buy a Teensy 3.2?

florind, 2020-03-23:
I have a couple of older boards and it's just enough for a project I'm doing. 
I tried to use both the SnoozeTimer and SnoozeDigital to wake the CPU back up. There are references to a low power mode in the datasheet but not much further. Anyway, thanks for trying to help.

closed 2020-03-23 by florind