duff2013/Snooze #79: Hibernate All Wakeups Teensy 3.2 & 3.6

ettare, 2020-04-12:
Apologies if this has already been covered, but I couldn't find a definitive answer:

Working on old computer: Arduino 1.8.9, Teensy Loader 1.46 , Snooze 6.3.2

Not working on new computer: Arduino 1.8.12, Teensy Loader 1.51 , Snooze 6.3.6

What is the latest combination of software that works please?

duff2013, 2020-04-12:
Can you download the latest one here. [https://github.com/duff2013/Snooze/releases/tag/v6.3.6](https://github.com/duff2013/Snooze/releases/tag/v6.3.6)

duff2013, 2020-04-12:
The latest should work can you explain what exactly you are seeing?

ettare, 2020-04-13:
With Arduino 1.8.12, Teensy Loader 1.51, and Snooze 6.3.6 (downloaded yesterday) , Teensys go to sleep, but don't wake up on timer.  I haven't tested other wakeup modes.  

Are there any tests I could make, or logs to send?  (Nothing too hard please - I'm just a noob...)

ettare, 2020-04-13:
I tested again today with  Arduino 1.8.12, Teensy Loader 1.51, and Snooze 6.3.6 and its working on both Teensys.

I think I must have had a confusion of Snooze versions due to copying some files from old to new computers - said I was a boob...

Looks like this issue can be closed.  Thanks for your prompt reply, and excellent software.

closed 2020-04-13 by ettare

ettare, 2020-04-13:
I found the reason for my problem.  The latest Teensyduino has an old version of Snooze; however even if you load a new version of Snooze (as a .Zip), Arduino still uses the old Teensyduino version.

The solution of course is to delete Snooze from:
C:\Program Files x86)\Arduino\hardware\teensy\avr\libraries
before you add the load the new version of Snooze.