duff2013/Snooze #80: Sleep working on Teensy LC

32-bits, 2020-04-29:
Hi Duff,

Thanks for writing such a useful libuary. I have migrated a project from a Teensy 3.6 where my code ran OK to a Teensy LC, to get better low current operation. Problem is I can not get your libuary to work on the LC. See my post on the Teensy forum https://forum.pjrc.com/threads/60676-Teensy-LC-not-resuming-from-snooze-(slleep) as all is explained there along with my simple code. Not sure if I have coded it wrong or find a bug and any assistance is welcome.

32bits

duff2013, 2020-04-29:
I haven't really looked at your code but it will not compile as is, i.e.`#include "project deffs.h"` and `#include "BLE remote.h"`. Also you need to point to the exact libraries you are using on GitHub, Teensy Libraries, etc..

Have you tried just seeing if the Timer driver wakes it without all the external libraries, can you modify one of the examples to just wake with the Timer driver?

32-bits, 2020-04-30:
I found the problem was due to installing the snooze lib via arduino as a zip file. I now load up visual code with platformio and use platformio facility to load via lib tab, it finds your lib and all works OK. My next problem is to get the current draw lower, at the moment it is 4.7ma in sleep mode. Thanks for your reply and writing the lib

duff2013, 2020-04-30:
Do you have anything connected to the Teensy that your sketch says you like BME device?

32-bits, 2020-05-01:
I have a BME280 sensore connected to the LC 3v3 output and a Bluetooth module also. I plan to insert a meter capable of reading ma currents inline with each device to see if they are the problem. If they are then using a spare LC pin to switch a FET or transistor to then supply the BME and Bluetooth module might work. This should then remove the supply line to the devices while in sleep.