duff2013/Snooze #84: Teensy LC: unsupported speeds 8 MHz and 2 MHz

XFer012, 2020-06-19:
Hello,
while Teensy 3.2 sketches with Snooze build OK down to 2 MHz clock, LC sketches only build down to 24 MHz.
I've enabled 8 MHz and 2 MHz in boards.txt according to a suggestion by Paul Stoffregen and get this error:

`In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src/hal/TEENSY_LC/hal.h:46:0,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src/Snooze.h:37,

                 from F:\Arduino\TeensyLowPowerTestOK\TeensyLowPowerTestOK.ino:1:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src/hal/TEENSY_LC/SnoozeUSBSerial.h: In member function 'SnoozeUSBSerial::operator bool()':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src/hal/TEENSY_LC/SnoozeUSBSerial.h:59:16: error: 'usb_configuration' was not declared in this scope

         return usb_configuration && ( usb_cdc_line_rtsdtr & USB_SERIAL_DTR ) && ( ( uint32_t )( systick_millis_count - usb_cdc_line_rtsdtr_millis ) >= 15 );

                ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src/hal/TEENSY_LC/SnoozeUSBSerial.h:59:39: error: 'usb_cdc_line_rtsdtr' was not declared in this scope

         return usb_configuration && ( usb_cdc_line_rtsdtr & USB_SERIAL_DTR ) && ( ( uint32_t )( systick_millis_count - usb_cdc_line_rtsdtr_millis ) >= 15 );

                                       ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src/hal/TEENSY_LC/SnoozeUSBSerial.h:59:61: error: 'USB_SERIAL_DTR' was not declared in this scope

         return usb_configuration && ( usb_cdc_line_rtsdtr & USB_SERIAL_DTR ) && ( ( uint32_t )( systick_millis_count - usb_cdc_line_rtsdtr_millis ) >= 15 );

                                                             ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src/hal/TEENSY_LC/SnoozeUSBSerial.h:59:120: error: 'usb_cdc_line_rtsdtr_millis' was not declared in this scope

         return usb_configuration && ( usb_cdc_line_rtsdtr & USB_SERIAL_DTR ) && ( ( uint32_t )( systick_millis_count - usb_cdc_line_rtsdtr_millis ) >= 15 );`

Looks like a few declarations are missing for 8 and 2 MHz.
Any hint to fix it myself?
Thanks!

duff2013, 2020-06-21:
This error is because when you build below 24 MHz "usb_configuration", "usb_cdc_line_rtsdtr", etc.. are not declared since the usb hardware won't work at those slow speeds. I'll fix this in the next update, I'm going to spend some time on Snooze, thanks for bringing this to my attention!

XFer012, 2020-06-21:
Thank you for your outstanding work! 
On Teensy 3.2 those vars are correctly defined even at 8 and 2 MHz, and I can confirm Snooze works OK at those speeds. :)
On the LC, 8 and 2 MHz are "unofficial" but they work correctly; it will be great to have them supported on the next Snooze version, thanks! :)

closed 2020-06-21 by XFer012