duff2013/Snooze #87: CAN-BUS

dufi2profor, 2020-07-11:
Hello,

i plan to use Teensy 4.0 as an gateway between two devices in my car.
I would like to ask, if You plan to implement CAN bus RX pin as an wake pin, or should i use digital pin driver?

Thank You

david-res, 2020-07-15:
Try to use `digital.pinMode(23, INPUT, RISING);` on your Rx pin (I use CAN1 on the T4)
This will work only in hibernate mode. Otherwise, you will need to reinitialise your CAN library upon wakeup from sleep or deepsleep mode.

Give that a try and report back.

dufi2profor, 2020-07-20:
david, thanks. I am waking Teensy by ERR output from TJA1055. That pin indicates start of frame on bus.
My sketch is based on bi directional forward example of FlexCAN library for T4.
Whenever valid message is received on "car" side, i preload one variable, that decrements every millisecond. Then, when car goes to sleep, the variable gets to zero, but before it does, i put all three TJA1055 to sleep and then, i call Snooze.deepSleep(config_teensy40). 
config_teensy40 has digital and usb loaded.
When "car side" TJA1055 indicates begin of transmission by pulling its ERR pin down, Teensy wakes up, enables all three TJA1055 and starts forwarding.
But sometimes Teensy gets frozen, in various times after wake up. It never freezes when You power it up. But only after wake.
The same happens sporadically when i configure additional pin and put button to it. Teensy wakes up and then freezes. Only way to get it run is to unplug power and plug it back again. 
Need to say that even when Teensy runs as it should, means does not freeze, when i press the button, it freezes at that moment.

I have used MsTimer2, which is ISR based to flash LED to see, if, when Teensy freezes, my program is not stuck somewhere. Both MsTimer2 stops.

By the way, when i tried hibernate sketch hibernate_all_wakeups, i am not able to wake the Teensy. So that is why i did not try hibernate, but deepSleep.

[PSA_GW_NAC_08A.txt](https://github.com/duff2013/Snooze/files/4949059/PSA_GW_NAC_08A.txt)

duff2013, 2020-07-20:
I'm a bit behind in these issues as of late, (kids school things). I'm hoping to get back to Snooze soon and I'll be taking a look at this.

dufi2profor, 2020-07-20:
Many thanks. Maybe if i call detachInterrupt on those two pins, it might help. Then reattach it before entering sleep.

dufi2profor, 2020-07-20:
I am using Arduino 1.8.9 and Teensyduino 1.5.3

dufi2profor, 2020-07-20:
Seems like this approach helped:

`      digital.pinMode(TJA_MASTER_ERR, INPUT, FALLING);      //TJA_MASTER_ERR, mode, type
      digital.pinMode(IN_DEBUG, INPUT_PULLUP, FALLING);      //TJA_MASTER_ERR, mode, type

      Snooze.deepSleep(config_teensy40);  // return module that woke processor

      detachInterrupt(digitalPinToInterrupt(TJA_MASTER_ERR));
      detachInterrupt(digitalPinToInterrupt(IN_DEBUG));`

david-res, 2020-07-20:
@dufi2profor I'm using the TI SN65HVD230, but I just had a look at the TJA1005 and it looks interesting - do you mind sharing a schematic of how you have it connected to the T4 with the wakeup and sleep functions?

dufi2profor, 2020-07-21:
Sure, i will draw the schema in KiCad and share then.
I have been using in past MCP2551, but that one, as part of high speed can group, has different recessive and dominant states.
As a result i found the output on RX pin inverted. Looked strange to me and then i looked in my radio (Peugeot 3008) and found this TJA1055. Low speed, fault tolerant. But different idle and dominant states.

I would not use control of sleep functions on those transcievers if i did not want the lowest current as possible and when all three TJAs are "sleeping", it makes current drop by at least 3 mA in total,

So my current setup, when using deepSleep draws 5 mA, which is okay in car i think.

david-res, 2020-07-21:
Actually, I just saw the 1055 limited to 125Kb/s baud rate, I'll stick with the SN65HVD230 in this case as I need 500Kb/s.
I put the 230 into sleep mode by default and pull the mode pin low when the Teensy is awake to bring it to life. I haven't measured consumption, but I'm hoping for 3-1mA if not less in Hibernate mode.

dufi2profor, 2020-07-24:
Yes, that is the only drawback of TJA, but i need it for my infotainment project in my Peugeot. Had been using that MCP2551, but only as listen only mode on my Peugeot "body" can which is running at 500 Kb/s.

Here is schematic diagram of what i have on breadboard now.

[SCH_CAN_ISOL_T4.pdf](https://github.com/duff2013/Snooze/files/4970745/SCH_CAN_ISOL_T4.pdf)