duff2013/Snooze #88: EEPROM.write hangs Teensy after wakeup

dufi2profor, 2020-08-04:
Hello,

i just found out that after wake up from deepSleep and using EEPRPOM.write(), Teensy hangs.

Here is example program, if You uncomment both EEPROM.write, code flashes the LED and Teensy gets to deepSleep properly.

Thank You very much
```
#include <EEPROM.h>

/***************************************
  This shows all the wakeups for deepSleep

  Supported Micros: T-LC/3.x/4.0
****************************************/
#include <Snooze.h>
// Load drivers
SnoozeDigital digital;
SnoozeUSBSerial usb;

SnoozeBlock config_teensy40(usb, digital);

void setup() {
  Serial.begin(115200);  
  pinMode(LED_BUILTIN, OUTPUT);
  digital.pinMode(3, INPUT_PULLUP, FALLING);//pin, mode, type
}

void loop() {
  int who;
  /********************************************************
    feed the sleep function its wakeup parameters. Then go
    to deepSleep.
  ********************************************************/

  who = Snooze.deepSleep( config_teensy40 );// return module that woke processor
  
  digitalWrite(LED_BUILTIN, HIGH);
  
  EEPROM.write(0, 128);
  EEPROM.write(1, 255);
  
  if (who == 3) { // pin wakeup source is its pin value
    for (int i = 0; i < 3; i++) {
      digitalWrite(LED_BUILTIN, LOW);
      delay(200);
      digitalWrite(LED_BUILTIN, HIGH);
      delay(200);
    }
  }
}
```

clovehitch, 2021-01-23:
Has anyone figured out a workaround for this?

dufi2profor, 2021-01-23:
Well, i used other approach. 
My system was supposed to store to eeprom several variables, but as mentioned, after waking it up from sleep, the eeprom writes hung the Teensy board. 
Since in my project i am using TJA1045 CAN transcievers, which also have wakeup pin used to turn on power supply for the system, so prior to going to sleep, i call function that writes variables to eeprom and then, after all CAN communication stops i call sequence which puts all TJAs (i am using three) to sleep. When TJAs enter sleep, all turn off their wakeup pin which turns off the power supply and thus whole system. 
As my "master" TJA is connected to VBAT in car, it recognised wakeup sequence on CAN bus and turns on my power supply.

That is how i do it :)