duff2013/Snooze #92: Sample Sketch compare.pinmode() for T_4.x

Defragster, 2020-08-30:
Glancing through the code to answer forum question in: \libraries\Snooze\examples\sleep\sleep_all_wakeups\sleep_all_wakeups.ino

where pin 11 is not a T_4 compare pin I see coded to pin 11:
`    Teensy 4.0
    Compare pins: 0, 1, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23

    Teensy 3.x
    Compare pins: 11,9,4

    Teensy LC
    Compare pins: 11
  ********************************************************/
  // trigger at threshold values greater than 1.65v
  //(pin, type, threshold(v))
  //compare.pinMode(11, HIGH, 1.65);

  // trigger at threshold values less than 1.65v
  //(pin, type, threshold(v))
  compare.pinMode(11, LOW, 1.65);
}`

duff2013, 2020-08-30:
Yes you are correct and I'll update the "compare" wakeup for the Teensy4, Thanks!

Defragster, 2020-08-30:
Glad I saw that in the right context.  Someone on the forum having issue with [https://forum.pjrc.com/threads/62763-Snooze-Calling-a-function-on-a-pin-interrupt](https://forum.pjrc.com/threads/62763-Snooze-Calling-a-function-on-a-pin-interrupt)

Way out of my experience with snooze - but seemed like a simple question to start. So scanned some code - but there is a lot behind the scenes ...

duff2013, 2020-08-30:
It is a lot of code that touches a lot of hardware on the Teensy's so it's been a journey trying to get everything to play nice with each other. The actual code to put a Teensy to sleep is not that bad for the Teensy 3.x not so much for the T4 but it's the wakeup code that gets hairy.

Defragster, 2020-08-30:
Saw your post on that thread - all the best on the family health front - your code does a great job encapsulating the transition and hardware with lots of details to handle. Waking and sleeping and and handling transitions across that for tracking interrupts with certain amounts of dead time in the process, and snooze needing to be the master to make it happen usably.