duff2013/Snooze #95: SnoozeDigital on pin #33-41 for T4.x

baichen, 2020-10-16:
Hi Duff, thank you for providing this great library. 

I was trying to use SnoozeDigital on pin # greater than 33 for Teensy 4.1 board, and realized that the pin number is limited to 32. And the wakeup codes for other SnoozeBlocks are hard-coded subsequently. 

I modified the source code to increase the hard-coded numbers accordingly. It seems that the SnoozeDigital is working on pin #33-35. I wonder if there's any potential issue with this modification.

Thank you very much!

duff2013, 2020-10-16:
As long as you change the hard coded numbers for the other non pin wakeups to a higher number, this is something I intend to do. Probably start the non pin wakeup at 100.

baichen, 2020-10-20:
Thank you very much!