duff2013/Snooze #101: Errors when compiling on TD1.54

david-res, 2021-07-07:
I am getting the following errors when I try to compile a sketch using snooze on the most recent release of Teensyduino 1.54
Compiling for Teensy 4.0

`/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp: In member function 'virtual void SnoozeDigital::enableDriver(uint8_t)':
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:12:15: error: expected unqualified-id before numeric constant
 #define DR    0
               ^
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy4/imxrt.h:5662:33: note: in expansion of macro 'DR'
 #define GPIO6_DR   (IMXRT_GPIO6.DR)
                                 ^
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:99:35: note: in expansion of macro 'GPIO6_DR'
                 case ( uint32_t )&GPIO6_DR: {
                                   ^
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:12:15: error: expected ')' before numeric constant
 #define DR    0
               ^
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy4/imxrt.h:5662:33: note: in expansion of macro 'DR'
 #define GPIO6_DR   (IMXRT_GPIO6.DR)
                                 ^
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:99:35: note: in expansion of macro 'GPIO6_DR'
                 case ( uint32_t )&GPIO6_DR: {
                                   ^
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:502:1: error: expected ':' at end of input
 }
 ^
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:502:1: error: expected statement at end of input
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:502:1: error: expected '}' at end of input
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:502:1: error: expected '}' at end of input
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:502:1: error: expected '}' at end of input
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp:502:1: error: expected '}' at end of input
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeCompare.cpp: In member function 'virtual void SnoozeCompare::disableDriver(uint8_t)':
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeCompare.cpp:174:18: warning: 'IRQ_CMP' may be used uninitialized in this function [-Wmaybe-uninitialized]
     IRQ_NUMBER_t IRQ_CMP;
                  ^
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeCompare.cpp: In member function 'virtual void SnoozeCompare::enableDriver(uint8_t)':
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Snooze/src/hal/TEENSY_40/SnoozeCompare.cpp:302:18: warning: 'IRQ_CMP' may be used uninitialized in this function [-Wmaybe-uninitialized]
     IRQ_NUMBER_t IRQ_CMP;
                  ^
Error compiling for board Teensy 4.0.`

CorBer, 2021-07-15:
Hi,

I just experienced the same issues also with for instance 1.54beta7.

regards
Cor

stefphd, 2021-08-03:
Hi,
I probably solved the same issue by commenting `#define DR 0` and replacing `IMR` with another name (e.g. `IMR0`). I do not known if it will work properly. Maybe the macros DR and IMR conflict with other macros defined in `teensy/avr/cores/teensy4/imxrt.h`.